iberianpig/fusuma #3: fusuma not working

set92, 2016-10-31:
I installed it via AUR https://aur.archlinux.org/packages/ruby-fusuma/ and I try to run it writing fusuma in the console, but it didn't work, and didnt show any trace of error, simply didnt do anything.

I tried to create the config file and put the example of Elementary and run it again but without luck.

I dont know if there is a debug mode option to run it and know which is the error.

diraimondo, 2016-10-31:
The same here: I think it gets the wrong input device. The absence of any option or debug mode didn't help: I discovered with a plain `ps aux | grep libinput`.

ewnd9, 2016-11-01:
For me, it works only with `sudo` even though I did `$ sudo gpasswd -a $USER input` as mentioned in README

Without sudo I see the same: no output, no errors

iberianpig, 2016-11-03:
Thanks.
I'll implement echo error messages when you don't get touch pad information.

diraimondo, 2016-11-03:
I would add an extensive, optional, debug output to the program. This could help to detect issues and would improve the reports to the project.

PR #9 (2016-11-14): Relax the device name regex
As event number can be more than single digit.

Probably closes #3.
closed 2016-11-14 by iberianpig