iberianpig/fusuma #15: sensitivity of 3 finger swipe

doubleloop, 2016-12-23:
Hi, thank you for the code. I like the idea.
I enabled 3finger swipe out/down but only ~30% of my swipes trigger expected action. I attach some example logs.

```
D, [2016-12-23T02:43:32.064241 #7348] DEBUG -- : {:libinput_command=>"stdbuf -oL -- libinput-debug-events --device       /dev/input/event1 --enable-dwt"}
D, [2016-12-23T02:43:32.955122 #7348] DEBUG -- : {:time=>"+0.89s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"7.60", :y=>"-78.90"}, :zoom=>"1.09"}}
D, [2016-12-23T02:43:32.981273 #7348] DEBUG -- : {:time=>"+0.92s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"7.25", :y=>"-117.65"}, :zoom=>"1.12"}}
D, [2016-12-23T02:43:33.006249 #7348] DEBUG -- : {:time=>"+0.94s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"16.00", :y=>"-128.76"}, :zoom=>"1.15"}}
D, [2016-12-23T02:43:33.030056 #7348] DEBUG -- : {:time=>"+0.96s", :action=>"GESTURE_PINCH_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2016-12-23T02:43:34.222167 #7348] DEBUG -- : {:time=>"+2.16s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"59.07"}, :zoom=>nil}}
D, [2016-12-23T02:43:34.248305 #7348] DEBUG -- : {:time=>"+2.18s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-21.10", :y=>"108.58"}, :zoom=>nil}}
D, [2016-12-23T02:43:34.273295 #7348] DEBUG -- : {:time=>"+2.21s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-27.00", :y=>"117.65"}, :zoom=>nil}}
D, [2016-12-23T02:43:34.298299 #7348] DEBUG -- : {:time=>"+2.23s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-21.00", :y=>"111.16"}, :zoom=>nil}}
D, [2016-12-23T02:43:34.323335 #7348] DEBUG -- : {:time=>"+2.26s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-14.00", :y=>"98.19"}, :zoom=>nil}}
D, [2016-12-23T02:43:34.373339 #7348] DEBUG -- : {:time=>"+2.31s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2016-12-23T02:43:35.021061 #7348] DEBUG -- : {:time=>"+2.96s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-4.87", :y=>"-74.75"}, :zoom=>nil}}
D, [2016-12-23T02:43:35.046028 #7348] DEBUG -- : {:time=>"+2.98s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-11.43", :y=>"-122.61"}, :zoom=>nil}}
D, [2016-12-23T02:43:35.071080 #7348] DEBUG -- : {:time=>"+3.00s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-6.25", :y=>"-153.77"}, :zoom=>nil}}
D, [2016-12-23T02:43:35.097222 #7348] DEBUG -- : {:time=>"+3.03s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-2.00", :y=>"-151.92"}, :zoom=>nil}}
D, [2016-12-23T02:43:35.122232 #7348] DEBUG -- : {:time=>"+3.06s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"-122.28"}, :zoom=>nil}}
D, [2016-12-23T02:43:35.122364 #7348] DEBUG -- : {:move=>#<struct Fusuma::ActionStack::Distance x=-4.91, y=-125.06599999999999>}
D, [2016-12-23T02:43:35.158020 #7348] DEBUG -- : {:time=>"+3.09s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2016-12-23T02:43:36.439502 #7348] DEBUG -- : {:time=>"+4.37s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-1.14", :y=>"44.94"}, :zoom=>nil}}
D, [2016-12-23T02:43:36.464454 #7348] DEBUG -- : {:time=>"+4.40s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-9.68", :y=>"54.92"}, :zoom=>nil}}
D, [2016-12-23T02:43:36.490648 #7348] DEBUG -- : {:time=>"+4.42s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-13.00", :y=>"71.33"}, :zoom=>nil}}
D, [2016-12-23T02:43:36.515620 #7348] DEBUG -- : {:time=>"+4.45s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-10.28", :y=>"61.50"}, :zoom=>nil}}
D, [2016-12-23T02:43:36.540616 #7348] DEBUG -- : {:time=>"+4.47s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-5.04", :y=>"44.35"}, :zoom=>nil}}
D, [2016-12-23T02:43:36.565618 #7348] DEBUG -- : {:time=>"+4.50s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-5.07", :y=>"47.01"}, :zoom=>nil}}
D, [2016-12-23T02:43:36.565696 #7348] DEBUG -- : {:move=>#<struct Fusuma::ActionStack::Distance x=-7.368333333333333, y=54.00833333333333>}
D, [2016-12-23T02:43:36.627589 #7348] DEBUG -- : {:time=>"+4.56s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2016-12-23T02:43:37.225354 #7348] DEBUG -- : {:time=>"+5.16s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.70", :y=>"-50.99"}, :zoom=>nil}}
D, [2016-12-23T02:43:37.250109 #7348] DEBUG -- : {:time=>"+5.18s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.85", :y=>"-75.23"}, :zoom=>nil}}
D, [2016-12-23T02:43:37.276280 #7348] DEBUG -- : {:time=>"+5.21s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"10.00", :y=>"-100.05"}, :zoom=>nil}}
D, [2016-12-23T02:43:37.301295 #7348] DEBUG -- : {:time=>"+5.24s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"7.00", :y=>"-85.22"}, :zoom=>nil}}
D, [2016-12-23T02:43:37.326620 #7348] DEBUG -- : {:time=>"+5.26s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"10.89", :y=>"-77.03"}, :zoom=>nil}}
D, [2016-12-23T02:43:37.363223 #7348] DEBUG -- : {:time=>"+5.30s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2016-12-23T02:43:38.226428 #7348] DEBUG -- : {:time=>"+6.16s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.86", :y=>"-80.56"}, :zoom=>"1.02"}}
D, [2016-12-23T02:43:38.251725 #7348] DEBUG -- : {:time=>"+6.19s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"7.71", :y=>"-133.03"}, :zoom=>"1.07"}}
D, [2016-12-23T02:43:38.276388 #7348] DEBUG -- : {:time=>"+6.21s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"16.00", :y=>"-119.50"}, :zoom=>"1.11"}}
D, [2016-12-23T02:43:38.313271 #7348] DEBUG -- : {:time=>"+6.25s", :action=>"GESTURE_PINCH_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2016-12-23T02:43:38.909942 #7348] DEBUG -- : {:time=>"+6.84s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-2.21", :y=>"-114.91"}, :zoom=>nil}}
D, [2016-12-23T02:43:38.936093 #7348] DEBUG -- : {:time=>"+6.87s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.25", :y=>"-179.66"}, :zoom=>nil}}
D, [2016-12-23T02:43:38.961100 #7348] DEBUG -- : {:time=>"+6.89s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"14.00", :y=>"-176.93"}, :zoom=>nil}}
D, [2016-12-23T02:43:38.984917 #7348] DEBUG -- : {:time=>"+6.92s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2016-12-23T02:43:39.531464 #7348] DEBUG -- : {:time=>"+7.47s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.25", :y=>"-112.90"}, :zoom=>nil}}
D, [2016-12-23T02:43:39.556470 #7348] DEBUG -- : {:time=>"+7.49s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"6.88", :y=>"-138.39"}, :zoom=>nil}}
D, [2016-12-23T02:43:39.581431 #7348] DEBUG -- : {:time=>"+7.52s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"8.00", :y=>"-122.28"}, :zoom=>nil}}
D, [2016-12-23T02:43:39.618361 #7348] DEBUG -- : {:time=>"+7.55s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2016-12-23T02:43:40.075593 #7348] DEBUG -- : {:time=>"+8.01s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-5.15", :y=>"-109.90"}, :zoom=>nil}}
D, [2016-12-23T02:43:40.100570 #7348] DEBUG -- : {:time=>"+8.04s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"-7.84", :y=>"-158.03"}, :zoom=>nil}}
D, [2016-12-23T02:43:40.126758 #7348] DEBUG -- : {:time=>"+8.06s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"6.25", :y=>"-186.20"}, :zoom=>nil}}
D, [2016-12-23T02:43:40.151754 #7348] DEBUG -- : {:time=>"+8.09s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"15.00", :y=>"-145.44"}, :zoom=>nil}}
[... cut ...]
D, [2016-12-23T02:43:41.824411 #7348] DEBUG -- : {:time=>"+9.76s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"37.00", :y=>"-128.76"}, :zoom=>nil}}
D, [2016-12-23T02:43:41.849414 #7348] DEBUG -- : {:time=>"+9.78s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"20.00", :y=>"-92.64"}, :zoom=>nil}}
D, [2016-12-23T02:43:41.849495 #7348] DEBUG -- : {:move=>#<struct Fusuma::ActionStack::Distance x=16.75, y=-111.52499999999999>}
D, [2016-12-23T02:43:41.883629 #7348] DEBUG -- : {:time=>"+9.79s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
```

martijnjanssen, 2016-12-23:
Exactly, I have been using fusuma for some while now. It is very good and usable but could improve on some points. I would really appreciate to be able to tweak the sensitivity of the gestures. It takes me some tries to get a gesture right at the moment due to the sensitivity. A scalar or a number for sesitivity would be great!

iberianpig, 2016-12-25:
OK, I will implement soon to change sensitivities with a config.

iberianpig, 2016-12-31:
I updated Fusuma implemented change sensitivity.
Please try `gem update fusuma`
and see README.

closed 2016-12-31 by iberianpig