iberianpig/fusuma #16: Best method to launch fusuma at start up

PrestanceDesign, 2016-12-24:
Hi,

Which is the best method to launch fusuma at start up ?

Ubuntu 16.10
/bin/zsh

Thx you for your work.
Merry christmas !

iberianpig, 2016-12-25:
If you are using ubuntu, you should try `gnome-session-properties`.
I'm using cerbere for registering as daemon, because my environment is elementary OS.

PrestanceDesign, 2016-12-26:
Yeah quite simply like xSwipe !
 I do not know where I had the head when I asked the question but I was absolutely looking to do it via upstart or something like that.

Thank you.

PrestanceDesign, 2017-01-03:
HNY !

So I tried `gnome-session-properties` but without success, maybe because I use `rvm` to manage my ruby environment.

I tried both in command input :
**Absolute path** : /home/user/.rvm/gems/ruby-2.3.3/bin/fusuma &
or directly fusuma.

Syslog line when log in :
`fusuma.desktop[4514]: /usr/bin/env: «ruby_executable_hooks»: Aucun fichier ou dossier de ce type`

iberianpig, 2017-01-03:
Sorry, I don't know about rvm.
where is your fusuma command?
your error message says 'fusuma correct path is missing'

Please check correct path with below command

```bash
$which fusuma
```

PrestanceDesign, 2017-01-03:
Thx for your helping.
As I say before, Fusuma path is : `/home/user/.rvm/gems/ruby-2.3.3/bin/fusuma`.
If I launch path in terminal, it works.

PrestanceDesign, 2017-01-03:
OK

closed 2017-01-03 by PrestanceDesign