iberianpig/fusuma #20: Not working for Ubuntu 16.10 with Gnome 3.20.4.

Whishay, 2017-01-09:
Thanks for the fantastic tools.

I'm using an Apple trackpad on Ubuntu 16.06 with gnome 3.20.4, does it support that? I followed the tips but get no output with debug mode but this. No fusuma like daemon after that  as well.

#fusuma -v
D, [2017-01-09T11:34:33.022116 #30109] DEBUG -- : {:libinput_command=>"stdbuf -oL -- libinput-debug-events --device /dev/input/#IO:0x0055c00220cac0 --enable-dwt"}

iberianpig, 2017-01-10:
@Whishay Could you paste a result of below command?

```
$libinput-list-devices
```

crespum, 2017-01-20:
@iberianpig it doesn't work to me neither. I'm using Debian Testing with Gome 3.22.2. If I use 4 finger gestures, the debug mode shows a 3 fingers gesture. If I use 3 or less fingers, the debug logs doesn't show anything.

This is the result of the command you are asking for:
```
Device:           SynPS/2 Synaptics TouchPad
Kernel:           /dev/input/event13
Group:            7
Seat:             seat0, default
Size:             69.68x50.22mm
Capabilities:     pointer 
Tap-to-click:     disabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *two-finger edge 
Click methods:    *button-areas clickfinger 
Disable-w-typing: enabled
Accel profiles:   none
Rotation:         n/a
```

iberianpig, 2017-01-21:
Hi, @crespum.

Thank you for sending the log.

Please update with `gem update fusuma`.

Please try below command If it doesn't work well with after update.

```
$ libinput-debug-events --device /dev/input/event13
```

Then please swipe with 3fingers on your touchpad.
Could you find the text like below in the output?

```
event6 	GESTURE_SWIPE_UPDATE +12.10s	3  7.25/ 0.00 (12.53/ 0.00 unaccelerated)
```

crespum, 2017-01-22:
@iberianpig, thanks for the help.

I have the last version installed (0.2.6).

If I use 2 or 3 fingers, the events are of type `POINTER_AXIS`. When I use 4 fingers, the event is `GESTURE_SWIPE_UPDATE`.

iberianpig, 2017-01-23:
@crespum Thank you.

This problem is caused by libinput.
Therefore, I can not help this issue.

We seem to need the function for finding out which actions are enabled in the debug mode.

closed 2017-01-24 by iberianpig

Jianzs, 2018-10-13:
> @crespum, thank you for sending the log.
>
> Please update with `gem update fusuma`.
>
> Please try below command If it doesn't work well with after update.
>
> ```
> $ libinput-debug-events --device [...]

Hi.The fusuma cannot work for my ubuntu 18.04. I tried the command. And when I use 3 or 4 fingers,no text appears.Can you help me? Thanks~