iberianpig/fusuma #21: Natural Scrolling

GeorgiosNikas, 2017-01-10:
Hi im using Ubuntu 16.10 with Gnome 3.20 and its working fine, but the scrolling direction is inverted. Under Settings the Option for Natural Scrolling disappeared. How do I enable it?

iberianpig, 2017-01-11:
If you want to set scroll direction as inverted, you should try to set your config file.
You can customize the settings for gestures to put and edit ~/.config/fusuma/config.yml.

GeorgiosNikas, 2017-01-11:
The problem is that i dont have the Fusuma folder in .config, I searched everywhere for it but I cant find it.

iberianpig, 2017-01-11:
Please try create fusuma directory in `~/.config` and put config.yml in fusuma directory.

GeorgiosNikas, 2017-01-11:
And what should I write into the file?

Saboteur777, 2017-01-14:
@GeorgiosNikas I did the following: 

1. enabled the natural scroll in Mouse and Keyboards > Natural scrolling 
2. then created ~/.config/fusuma folder
3. created config.yml, with the following content: 

```
swipe:
  3: 
    left: 
      shortcut: 'alt+Right'
    right: 
      shortcut: 'alt+Left'
    #up: 
    #  shortcut: ''
    down: 
      shortcut: 'ctrl+r'
  4:
    left: 
      shortcut: 'ctrl+alt+Right'
    right: 
      shortcut: 'ctrl+alt+Left'
    up: 
      shortcut: 'ctrl+alt+Down'
    down: 
      shortcut: 'ctrl+alt+Up'

threshold:
  swipe: 0.15
  pinch: 0

```

You can customise it as you wish from now. **Note:** I disabled pinch-to-zoom. Lines starting with _#_ are commented out.

GeorgiosNikas, 2017-01-14:
The natural scrolling option disappeared from the settings, the only setting left is for the primary key. I also edited the config.yml but nothing happens.

Saboteur777, 2017-01-14:
Have you uninstalled one or more packages from your system? If you don't run fusuma, the natural scrolling option is visible?

iberianpig, 2017-01-21:
If you have any problem, please reopen this issue.
Thank you.

closed 2017-01-21 by iberianpig