iberianpig/fusuma #22: Fusuma doing nothing on ubuntu 16.04

prasantadh, 2017-01-11:
I am using ubuntu 16.04 on Fusuma isn't doing anything at the moment at all.

The output of `fusuma -v` is as follows:
`D, [2017-01-11T06:22:45.416046 #31763] DEBUG -- : Device:           Power Button

D, [2017-01-11T06:22:45.416146 #31763] DEBUG -- : Kernel:           /dev/input/event3

D, [2017-01-11T06:22:45.416224 #31763] DEBUG -- : Group:            1

D, [2017-01-11T06:22:45.416252 #31763] DEBUG -- : Seat:             seat0, default

D, [2017-01-11T06:22:45.416269 #31763] DEBUG -- : Capabilities:     keyboard 

D, [2017-01-11T06:22:45.416283 #31763] DEBUG -- : Tap-to-click:     n/a

D, [2017-01-11T06:22:45.416303 #31763] DEBUG -- : Tap-and-drag:     n/a

D, [2017-01-11T06:22:45.416317 #31763] DEBUG -- : Tap drag lock:    n/a

D, [2017-01-11T06:22:45.416329 #31763] DEBUG -- : Left-handed:      n/a

D, [2017-01-11T06:22:45.416342 #31763] DEBUG -- : Nat.scrolling:    n/a

D, [2017-01-11T06:22:45.416354 #31763] DEBUG -- : Middle emulation: n/a

D, [2017-01-11T06:22:45.416366 #31763] DEBUG -- : Calibration:      n/a

D, [2017-01-11T06:22:45.416379 #31763] DEBUG -- : Scroll methods:   none

D, [2017-01-11T06:22:45.416391 #31763] DEBUG -- : Click methods:    none

D, [2017-01-11T06:22:45.416403 #31763] DEBUG -- : Disable-w-typing: n/a

D, [2017-01-11T06:22:45.416415 #31763] DEBUG -- : Accel profiles:   n/a

D, [2017-01-11T06:22:45.416427 #31763] DEBUG -- : 

D, [2017-01-11T06:22:45.416440 #31763] DEBUG -- : Device:           Video Bus

D, [2017-01-11T06:22:45.416452 #31763] DEBUG -- : Kernel:           /dev/input/event5

D, [2017-01-11T06:22:45.416468 #31763] DEBUG -- : Group:            2

D, [2017-01-11T06:22:45.416481 #31763] DEBUG -- : Seat:             seat0, default

D, [2017-01-11T06:22:45.416494 #31763] DEBUG -- : Capabilities:     keyboard 

D, [2017-01-11T06:22:45.416506 #31763] DEBUG -- : Tap-to-click:     n/a

D, [2017-01-11T06:22:45.416520 #31763] DEBUG -- : Tap-and-drag:     n/a

D, [2017-01-11T06:22:45.416535 #31763] DEBUG -- : Tap drag lock:    n/a

D, [2017-01-11T06:22:45.416547 #31763] DEBUG -- : Left-handed:      n/a

D, [2017-01-11T06:22:45.416560 #31763] DEBUG -- : Nat.scrolling:    n/a

D, [2017-01-11T06:22:45.416572 #31763] DEBUG -- : Middle emulation: n/a

D, [2017-01-11T06:22:45.416584 #31763] DEBUG -- : Calibration:      n/a

D, [2017-01-11T06:22:45.416596 #31763] DEBUG -- : Scroll methods:   none

D, [2017-01-11T06:22:45.416608 #31763] DEBUG -- : Click methods:    none

D, [2017-01-11T06:22:45.416621 #31763] DEBUG -- : Disable-w-typing: n/a

D, [2017-01-11T06:22:45.416633 #31763] DEBUG -- : Accel profiles:   n/a

D, [2017-01-11T06:22:45.416645 #31763] DEBUG -- : 
[... 274 lines cut ...]
Disable-w-typing: enabled
Accel profiles:   none

Device:           Dell WMI hotkeys
Kernel:           /dev/input/event14
Group:            8
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a`

And `ps aux | grep libinput` I am not sure if this one helps but I saw it was a part of debug process in one of the issues.
`p  31767  0.1  0.0  34656  1736 pts/6    S+   06:22   0:00 libinput-debug-events --device /dev/input/event6 --enable-dwt
p  32194  0.0  0.0  14224  1092 pts/2    S+   06:33   0:00 grep --color=auto libinput`

Thank you!

iberianpig, 2017-01-11:
please confirm log with below command

```
$ libinput-debug-events --device /dev/input/event6
```

Then please swipe on your touchpad.
Could you find the text like below in the output?

```
event6 	GESTURE_SWIPE_UPDATE +12.10s	4  7.25/ 0.00 (12.53/ 0.00 unaccelerated)
```

prasantadh, 2017-01-11:
The command `libinput-debug-events --device /dev/input/event6` gives me first
`event6 	DEVICE_ADDED     AlpsPS/2 ALPS GlidePoint          seat0 default group1 cap:p	size 95.24/43.75mm tap(dl off) left scroll-nat scroll-2fg-edge dwt-on`

 and then on swipe something similar to:
`event6 	POINTER_MOTION    +1.47s	 -0.06/  0.00`
or 
`event6 	POINTER_AXIS      +6.45s	vert 2.71* horiz 0.00`
or
`libinput error: libinput bug: Event for missing capability CAP_GESTURE on device "AlpsPS/2 ALPS GlidePoint"`

The first one was for single finger gesture, the second one is for two finger gesture and the third one is for a three or four finger gesture. I am essentially trying to emulate some mac-like behavior with ubuntu. I am using unity desktop environment and have confirmed that thee trackpad supports up to 4 finger gestures with `geisview`.

iberianpig, 2017-01-16:
It seems not to able to catch 3/4  fingers swipe event with libinput driver.
This is libinput driver's issue.
So that, you can't use fusuma even if libinput driver supports 1/2 finger swipe with your touchpad.

closed 2017-01-16 by iberianpig