iberianpig/fusuma #23: Not Working in ElementaryOS Loki

kenedyroger, 2017-01-18:
Hello!
Can It works with ElementaryOS Loki and touchpad "ETPS/2 Elantech Touchpad"?

I follow README and not get errors but any gestures works.
PS: On Windows7 my touchpad works multigestures.

*Sorry if I wrote wrong text. I'm not English native.

iberianpig, 2017-01-20:
please confirm the log with below command

```
$ libinput-debug-events
```

Then please swipe on your touchpad.
Could you find the text like below in the output?

```
event6 	GESTURE_SWIPE_UPDATE +12.10s	4  7.25/ 0.00 (12.53/ 0.00 unaccelerated)
```

kenedyroger, 2017-01-20:
When I use touch the message log shows:

libinput error: libinput bug: Event for missing capability CAP_GESTURE on
device "ETPS/2 Elantech Touchpad"

iberianpig, 2017-01-21:
This is the issue of libinput driver, therefore Fusuma can't recognize your gesture.

closed 2017-01-21 by iberianpig

get4prince, 2018-11-19:
hey there my  2 pinch is working fine but the problem is with 3 and 4 finger . 
in 3 and 4 finger in get same event6   POINTER_AXIS