iberianpig/fusuma #24: Could u add a "-c" options to specify config file manually?

liyaoxinchifan, 2017-01-21:
I think fusuma is a nice software enhance my touchpad, and I wanna run fusuma on boot with systemd, I found:
```
def file_path
    filename = 'fusuma/config.yml'
    original_path = File.expand_path "~/.config/#{filename}"
    default_path  = File.expand_path "../../#{filename}", __FILE__
    File.exist?(original_path) ? original_path : default_path
end
```
in config.rb, so I specify user and $HOME in systemd config file like:
```
[Service]
User=leeyx
Environment=HOME=/home/leeyx
...
...
...
```
But fusuma seems can't load config.yml by use ` "~/.config/#{filename}"`
Could you add a options to specify config file directly?

Thanks a lot!

iberianpig, 2017-01-22:
Hi, @liyaoxinchifan.
Thank you for your suggestion.
I will implement this feature soon.

iberianpig, 2017-01-22:
@liyaoxinchifan Updated!
Please update with `$ gem update fusuma`
You can use `-c` option, `fusuma --c path/to/custom_path.yml`.

closed 2017-01-22 by iberianpig