iberianpig/fusuma #26: Error after upgrading to libinput 1.6.0

tobac, 2017-01-24:
The Arch Linux libinput package maintainers now package libinput 1.6.0, which leaves me with an error when running fusuma and trying to swipe. It did work fine on 1.5.4 (the previous version under Arch).

`/home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/lib/fusuma/gesture_action.rb:46:in `parse_finger_directions': undefined method `tr' for nil:NilClass (NoMethodError)
        from /home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/lib/fusuma/gesture_action.rb:33:in `gesture_action_arguments'
        from /home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/lib/fusuma/gesture_action.rb:22:in `initialize_by'
        from /home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:38:in `block (2 levels) in read_libinput'
        from /home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:37:in `each'
        from /home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:37:in `block in read_libinput'
        from /usr/lib/ruby/2.4.0/open3.rb:205:in `popen_run'
        from /usr/lib/ruby/2.4.0/open3.rb:95:in `popen3'
        from /home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:36:in `read_libinput'
        from /home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:21:in `run'
        from /home/corn/.gem/ruby/2.4.0/gems/fusuma-0.2.6/exe/fusuma:21:in `<top (required)>'
        from /home/corn/.gem/ruby/2.4.0/bin/fusuma:22:in `load'
        from /home/corn/.gem/ruby/2.4.0/bin/fusuma:22:in `<main>'`

iberianpig, 2017-01-25:
@tobac 
I can't apply libinput v1.6.0 on my environment(elementary OS) now.
So, I can't debug it.
Could you show me the output of `libinput-debug-events` with some swipe gestures?

tobac, 2017-01-25:
Sure. Here are some four- and three-finger swipes:

```
 event10  GESTURE_SWIPE_BEGIN  +9.53s   4
 event10  GESTURE_SWIPE_UPDATE  +9.53s  4  0.00/26.33 ( 0.00/53.60 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +9.54s  4 -0.08/23.39 (-0.21/62.16 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +9.55s  4 -6.38/34.75 (-16.86/91.79 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +9.55s  4 -5.88/60.17 (-9.11/93.18 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +9.56s  4 -10.79/66.69 (-16.23/100.35 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +9.57s  4 -12.10/92.78 (-14.55/111.58 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +9.58s  4 -5.39/87.69 (-6.39/103.88 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +9.59s  4 -12.61/106.17 (-14.03/118.11 unaccelerated)
 event10  GESTURE_SWIPE_END  +9.64s     4
 event10  GESTURE_SWIPE_BEGIN +10.30s   4
 event10  GESTURE_SWIPE_UPDATE +10.30s  4 69.16/-7.32 (103.14/-10.91 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +10.31s  4 35.50/-2.46 (81.88/-5.67 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +10.32s  4 51.54/-3.78 (94.87/-6.95 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +10.33s  4 69.37/ 0.89 (99.68/ 1.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +10.34s  4 88.20/-4.76 (112.98/-6.10 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +10.34s  4 109.17/ 3.99 (117.17/ 4.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +10.35s  4 213.50/-1.33 (172.03/-1.07 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +10.36s  4 296.33/ 0.18 (173.19/ 0.11 unaccelerated)
 event10  GESTURE_SWIPE_END +10.39s     4
 event10  GESTURE_SWIPE_BEGIN +12.38s   3
 event10  GESTURE_SWIPE_UPDATE +12.38s  3 -77.21/ 0.00 (-67.99/ 0.00 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.39s  3 -25.83/ 0.00 (-69.81/ 0.00 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.39s  3 -29.35/ 0.37 (-79.30/ 1.00 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.40s  3 -38.86/11.61 (-87.81/26.25 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.41s  3 -54.79/13.63 (-96.33/23.96 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.42s  3 -115.78/38.10 (-131.79/43.36 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.43s  3 -300.70/94.31 (-194.20/60.91 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.44s  3 -241.47/74.74 (-147.01/45.50 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.44s  3 -236.32/87.67 (-162.65/60.34 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.45s  3 -262.58/107.91 (-164.88/67.76 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.46s  3 -258.94/118.73 (-160.83/73.75 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +12.47s  3 -203.36/85.78 (-139.33/58.77 unaccelerated)
 event10  GESTURE_SWIPE_END +12.49s     3
 event10  GESTURE_SWIPE_BEGIN +13.12s   3
 event10  GESTURE_SWIPE_UPDATE +13.12s  3 55.05/-7.57 (81.95/-11.27 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.13s  3 28.06/-4.33 (75.81/-11.70 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.13s  3 46.91/-9.77 (97.31/-20.26 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.14s  3 70.24/-10.37 (105.27/-15.55 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.15s  3 80.29/-11.85 (110.15/-16.26 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.16s  3 205.06/-31.52 (173.54/-26.67 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.16s  3 182.88/-20.48 (135.00/-15.12 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.17s  3 291.46/-19.44 (196.71/-13.12 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.18s  3 201.99/-1.91 (136.12/-1.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.19s  3 233.06/-4.20 (173.95/-3.14 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.20s  3 267.95/-3.82 (170.05/-2.42 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.21s  3 298.13/ 9.52 (183.17/ 5.85 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.21s  3 179.62/ 3.11 (131.79/ 2.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.22s  3 104.86/ 3.43 (113.22/ 3.71 unaccelerated)
 event10  GESTURE_SWIPE_END +13.24s     3
 event10  POINTER_MOTION   +13.62s       -1.20/  0.00
 event10  GESTURE_SWIPE_BEGIN +13.69s   3
 event10  GESTURE_SWIPE_UPDATE +13.69s  3  0.00/-14.64 ( 0.00/-43.51 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.70s  3  0.98/-19.69 ( 2.65/-53.21 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.71s  3  5.22/-27.98 (14.10/-75.60 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.72s  3  3.69/-36.50 ( 8.94/-88.30 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.73s  3 21.32/-89.51 (29.32/-123.10 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.73s  3 21.65/-122.01 (22.20/-125.10 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.74s  3 40.03/-158.23 (36.16/-142.93 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.75s  3 48.87/-213.52 (36.44/-159.19 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.76s  3 68.58/-290.09 (42.86/-181.30 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +13.77s  3 39.09/-241.22 (25.13/-155.06 unaccelerated)
 event10  GESTURE_SWIPE_END +13.79s     3
 event10  GESTURE_SWIPE_BEGIN +14.18s   3
 event10  GESTURE_SWIPE_UPDATE +14.18s  3 -6.60/27.23 (-11.73/48.36 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.19s  3 -12.87/22.17 (-34.76/59.91 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.19s  3 -5.09/32.49 (-13.26/84.73 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.20s  3 -37.02/100.05 (-47.61/128.67 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.21s  3 -42.59/234.43 (-31.41/172.89 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.22s  3 -52.84/311.85 (-31.27/184.58 unaccelerated)
[... cut ...]
 event10  GESTURE_SWIPE_UPDATE +14.24s  3 -19.71/357.04 (-11.45/207.41 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.24s  3 -12.03/347.51 (-6.42/185.44 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.25s  3 -10.25/369.19 (-5.58/201.13 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.26s  3 -6.51/291.00 (-3.77/168.61 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.27s  3 -2.03/294.46 (-1.26/182.59 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE +14.28s  3  9.11/167.56 ( 6.84/125.81 unaccelerated)
 event10  GESTURE_SWIPE_END +14.28s     3
```

Saboteur777, 2017-01-28:
OpenSUSE Tumbleweed, libinput-tools 1.6.0-1.1, might be related: 

after fusuma -v: 

```
~$ fusuma -v                  
D, [2017-01-28T12:15:27.358245 #16600] DEBUG -- : Device:           Power Button

D, [2017-01-28T12:15:27.358494 #16600] DEBUG -- : Kernel:           /dev/input/event5

D, [2017-01-28T12:15:27.358612 #16600] DEBUG -- : Group:            1

D, [2017-01-28T12:15:27.358671 #16600] DEBUG -- : Seat:             seat0, default

D, [2017-01-28T12:15:27.358724 #16600] DEBUG -- : Capabilities:     keyboard 

D, [2017-01-28T12:15:27.358765 #16600] DEBUG -- : Tap-to-click:     n/a

D, [2017-01-28T12:15:27.358817 #16600] DEBUG -- : Tap-and-drag:     n/a

D, [2017-01-28T12:15:27.358861 #16600] DEBUG -- : Tap drag lock:    n/a

D, [2017-01-28T12:15:27.358900 #16600] DEBUG -- : Left-handed:      n/a

D, [2017-01-28T12:15:27.358940 #16600] DEBUG -- : Nat.scrolling:    n/a

D, [2017-01-28T12:15:27.358978 #16600] DEBUG -- : Middle emulation: n/a

D, [2017-01-28T12:15:27.359016 #16600] DEBUG -- : Calibration:      n/a

D, [2017-01-28T12:15:27.359054 #16600] DEBUG -- : Scroll methods:   none

D, [2017-01-28T12:15:27.359093 #16600] DEBUG -- : Click methods:    none

D, [2017-01-28T12:15:27.359130 #16600] DEBUG -- : Disable-w-typing: n/a

D, [2017-01-28T12:15:27.359168 #16600] DEBUG -- : Accel profiles:   n/a

D, [2017-01-28T12:15:27.359205 #16600] DEBUG -- : Rotation:         n/a

D, [2017-01-28T12:15:27.359243 #16600] DEBUG -- : 

D, [2017-01-28T12:15:27.359282 #16600] DEBUG -- : Device:           Video Bus

D, [2017-01-28T12:15:27.359321 #16600] DEBUG -- : Kernel:           /dev/input/event6

D, [2017-01-28T12:15:27.359376 #16600] DEBUG -- : Group:            2

D, [2017-01-28T12:15:27.359418 #16600] DEBUG -- : Seat:             seat0, default

D, [2017-01-28T12:15:27.359457 #16600] DEBUG -- : Capabilities:     keyboard 

D, [2017-01-28T12:15:27.359494 #16600] DEBUG -- : Tap-to-click:     n/a

D, [2017-01-28T12:15:27.359592 #16600] DEBUG -- : Tap-and-drag:     n/a

D, [2017-01-28T12:15:27.359633 #16600] DEBUG -- : Tap drag lock:    n/a

D, [2017-01-28T12:15:27.359671 #16600] DEBUG -- : Left-handed:      n/a

D, [2017-01-28T12:15:27.359709 #16600] DEBUG -- : Nat.scrolling:    n/a

D, [2017-01-28T12:15:27.359747 #16600] DEBUG -- : Middle emulation: n/a

D, [2017-01-28T12:15:27.359784 #16600] DEBUG -- : Calibration:      n/a

D, [2017-01-28T12:15:27.359821 #16600] DEBUG -- : Scroll methods:   none

D, [2017-01-28T12:15:27.359859 #16600] DEBUG -- : Click methods:    none

[... 193 lines cut ...]

D, [2017-01-28T12:15:27.363785 #16600] DEBUG -- : Group:            7

D, [2017-01-28T12:15:27.363824 #16600] DEBUG -- : Seat:             seat0, default

D, [2017-01-28T12:15:27.363863 #16600] DEBUG -- : Size:             102.33x77.33mm

D, [2017-01-28T12:15:27.363902 #16600] DEBUG -- : Capabilities:     pointer 

D, [2017-01-28T12:15:27.363948 #16600] DEBUG -- : Tap-to-click:     disabled

D, [2017-01-28T12:15:27.553908 #16600] DEBUG -- : {:libinput_command=>"stdbuf -oL -- libinput-debug-events --device       /dev/input/event15 --enable-dwt"}
^C/home/saboteur/.rvm/gems/ruby-2.3.1/gems/fusuma-0.2.3/lib/fusuma.rb:35:in `each': Interrupt
        from /home/saboteur/.rvm/gems/ruby-2.3.1/gems/fusuma-0.2.3/lib/fusuma.rb:35:in `block in read_libinput'
        from /home/saboteur/.rvm/rubies/ruby-2.3.1/lib/ruby/2.3.0/open3.rb:205:in `popen_run'
        from /home/saboteur/.rvm/rubies/ruby-2.3.1/lib/ruby/2.3.0/open3.rb:95:in `popen3'
        from /home/saboteur/.rvm/gems/ruby-2.3.1/gems/fusuma-0.2.3/lib/fusuma.rb:34:in `read_libinput'
        from /home/saboteur/.rvm/gems/ruby-2.3.1/gems/fusuma-0.2.3/lib/fusuma.rb:24:in `run'
        from /home/saboteur/.rvm/gems/ruby-2.3.1/gems/fusuma-0.2.3/exe/fusuma:16:in `<top (required)>'
        from /home/saboteur/.rvm/gems/ruby-2.3.1/bin/fusuma:23:in `load'
        from /home/saboteur/.rvm/gems/ruby-2.3.1/bin/fusuma:23:in `<main>'
        from /home/saboteur/.rvm/gems/ruby-2.3.1/bin/ruby_executable_hooks:15:in `eval'
        from /home/saboteur/.rvm/gems/ruby-2.3.1/bin/ruby_executable_hooks:15:in `<main>'

```

peyloride, 2017-02-03:
Downgraded to 1.5.4 version from https://archive.archlinux.org/packages/l/libinput/ and it is working good now.

iberianpig, 2017-02-04:
I updated parser. Please update with `gem update fusuma`.

tobac, 2017-02-04:
Lovely! It's working, thanks a lot!

closed 2017-02-04 by tobac

patrick-ebot7, 2019-11-06:
This bug appears to be back:

```
fusuma -v
I, [2019-11-06T09:25:13.918176 #4953]  INFO -- : reload config : /home/<user>/.config/fusuma/config.yml
I, [2019-11-06T09:25:13.918225 #4953]  INFO -- : ---------------------------------------------
I, [2019-11-06T09:25:13.918237 #4953]  INFO -- : Fusuma: 1.2.1
I, [2019-11-06T09:25:13.918991 #4953]  INFO -- : libinput: 1.14.3
I, [2019-11-06T09:25:13.919547 #4953]  INFO -- : OS: Linux 5.2.21-1-MANJARO #1 SMP PREEMPT Sat Oct 12 11:09:56 UTC 2019
I, [2019-11-06T09:25:13.920001 #4953]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2019-11-06T09:25:13.922112 #4953]  INFO -- : Desktop session: gnome
I, [2019-11-06T09:25:13.922134 #4953]  INFO -- : ---------------------------------------------
D, [2019-11-06T09:25:13.922148 #4953] DEBUG -- : ---------------------------------------------
D, [2019-11-06T09:25:13.922156 #4953] DEBUG -- : Enabled Plugins: 
D, [2019-11-06T09:25:13.922200 #4953] DEBUG -- :   Fusuma::Plugin::Buffers::GestureBuffer
D, [2019-11-06T09:25:13.922211 #4953] DEBUG -- :   Fusuma::Plugin::Detectors::PinchDetector
D, [2019-11-06T09:25:13.922218 #4953] DEBUG -- :   Fusuma::Plugin::Detectors::RotateDetector
D, [2019-11-06T09:25:13.922233 #4953] DEBUG -- :   Fusuma::Plugin::Detectors::SwipeDetector
D, [2019-11-06T09:25:13.922280 #4953] DEBUG -- :   Fusuma::Plugin::Events::Records::GestureRecord
D, [2019-11-06T09:25:13.922296 #4953] DEBUG -- :   Fusuma::Plugin::Events::Records::IndexRecord
D, [2019-11-06T09:25:13.922305 #4953] DEBUG -- :   Fusuma::Plugin::Events::Records::TextRecord
D, [2019-11-06T09:25:13.922313 #4953] DEBUG -- :   Fusuma::Plugin::Executors::CommandExecutor
D, [2019-11-06T09:25:13.922321 #4953] DEBUG -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
D, [2019-11-06T09:25:13.922342 #4953] DEBUG -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
D, [2019-11-06T09:25:13.922349 #4953] DEBUG -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
D, [2019-11-06T09:25:13.922356 #4953] DEBUG -- : ---------------------------------------------
D, [2019-11-06T09:25:13.922985 #4953] DEBUG -- : {:debug_events=>"stdbuf -oL -- libinput debug-events"}
D, [2019-11-06T09:25:13.933871 #4953] DEBUG -- : {:input_event=>time: 2019-11-06 09:25:13 +0100, tag: libinput_command_input, record: event7  - Raydium Touchscreen: kernel bug: kernel has only x or y resolution, not both.}
D, [2019-11-06T09:25:13.934859 #4953] DEBUG -- : {:list_devices=>"libinput list-devices"}
Traceback (most recent call last):
        29: from /usr/bin/fusuma:23:in `<main>'
        28: from /usr/bin/fusuma:23:in `load'
        27: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/exe/fusuma:41:in `<top (required)>'
        26: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma.rb:18:in `run'
        25: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma.rb:95:in `run'
        24: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:20:in `run'
        23: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:55:in `debug_events'
        22: from /usr/lib/ruby/2.6.0/open3.rb:101:in `popen3'
        21: from /usr/lib/ruby/2.6.0/open3.rb:219:in `popen_run'
        20: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:56:in `block in debug_events'
        19: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:56:in `each'
        18: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:57:in `block (2 levels) in debug_events'
        17: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:21:in `block in run'
        16: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma.rb:96:in `block in run'
        15: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma.rb:105:in `filter'
        14: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma.rb:105:in `reduce'
        13: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma.rb:105:in `each'
        12: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma.rb:105:in `block in filter'
        11: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/filters/filter.rb:15:in `filter'
        10: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/filters/filter.rb:15:in `tap'
         9: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/filters/filter.rb:17:in `block in filter'
         8: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/filters/libinput_device_filter.rb:29:in `keep?'
         7: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/plugin/filters/libinput_device_filter.rb:54:in `all'
         6: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/device.rb:39:in `available'
         5: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/device.rb:33:in `all'
         4: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/device.rb:62:in `fetch_devices'
         3: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/device.rb:81:in `generate_devices'
         2: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/device.rb:81:in `each_with_object'
         1: from /usr/lib/ruby/gems/2.6.0/gems/fusuma-1.2.1/lib/fusuma/device.rb:81:in `each'
[... cut ...]
    left:
      command: 'xdotool key ctrl+Page_Down' 
    right:
      command: 'xdotool key ctrl+Page_Up'
    up:
      command: 'xdotool key ctrl+space'
    down:
      command: 'xdotool key ctrl+space'
  4:
    up:
      command: 'xdotool key super+m'
    down:
      command: 'xdotool key super+m'
pinch:
  in:
    command: 'xdotool key ctrl+='
    threshold: 0.1
  out:
     command: 'xdotool key ctrl+-'
     threshold: 0.1

threshold:
  swipe: 1
  pinch: 1

interval:
  swipe: 1
  pinch: 1
```

iberianpig, 2019-11-06:
@patrick-ebot7  Thank you for reporting!
I'll reproduce it with libinput 1.14.3.

patrick-ebot7, 2019-11-12:
Nothing has changed in my configuration, but everything is working again after a reboot yesterday.