iberianpig/fusuma #28: fusuma not working

crpig, 2017-01-28:
Here is the log output, can you help figure out what's wrong with the environment?

D, [2017-01-28T11:00:42.191649 #4407] DEBUG -- : Device:           Power Button

D, [2017-01-28T11:00:42.192361 #4407] DEBUG -- : Kernel:           /dev/input/event2

D, [2017-01-28T11:00:42.192615 #4407] DEBUG -- : Group:            1

D, [2017-01-28T11:00:42.192729 #4407] DEBUG -- : Seat:             seat0, default

D, [2017-01-28T11:00:42.192836 #4407] DEBUG -- : Capabilities:     keyboard 

D, [2017-01-28T11:00:42.192922 #4407] DEBUG -- : Tap-to-click:     n/a

D, [2017-01-28T11:00:42.193029 #4407] DEBUG -- : Tap-and-drag:     n/a

D, [2017-01-28T11:00:42.193111 #4407] DEBUG -- : Tap drag lock:    n/a

D, [2017-01-28T11:00:42.193193 #4407] DEBUG -- : Left-handed:      n/a

D, [2017-01-28T11:00:42.193281 #4407] DEBUG -- : Nat.scrolling:    n/a

D, [2017-01-28T11:00:42.193366 #4407] DEBUG -- : Middle emulation: n/a

D, [2017-01-28T11:00:42.193445 #4407] DEBUG -- : Calibration:      n/a

D, [2017-01-28T11:00:42.193524 #4407] DEBUG -- : Scroll methods:   none

D, [2017-01-28T11:00:42.193602 #4407] DEBUG -- : Click methods:    none

D, [2017-01-28T11:00:42.193681 #4407] DEBUG -- : Disable-w-typing: n/a

D, [2017-01-28T11:00:42.193759 #4407] DEBUG -- : Accel profiles:   n/a

D, [2017-01-28T11:00:42.193836 #4407] DEBUG -- : 

D, [2017-01-28T11:00:42.193917 #4407] DEBUG -- : Device:           Video Bus

D, [2017-01-28T11:00:42.194005 #4407] DEBUG -- : Kernel:           /dev/input/event4

D, [2017-01-28T11:00:42.194099 #4407] DEBUG -- : Group:            2

D, [2017-01-28T11:00:42.194181 #4407] DEBUG -- : Seat:             seat0, default

D, [2017-01-28T11:00:42.194263 #4407] DEBUG -- : Capabilities:     keyboard 

D, [2017-01-28T11:00:42.194343 #4407] DEBUG -- : Tap-to-click:     n/a

D, [2017-01-28T11:00:42.194428 #4407] DEBUG -- : Tap-and-drag:     n/a

D, [2017-01-28T11:00:42.194507 #4407] DEBUG -- : Tap drag lock:    n/a

D, [2017-01-28T11:00:42.194584 #4407] DEBUG -- : Left-handed:      n/a

D, [2017-01-28T11:00:42.195145 #4407] DEBUG -- : Nat.scrolling:    n/a

D, [2017-01-28T11:00:42.195243 #4407] DEBUG -- : Middle emulation: n/a

D, [2017-01-28T11:00:42.195325 #4407] DEBUG -- : Calibration:      n/a

D, [2017-01-28T11:00:42.195402 #4407] DEBUG -- : Scroll methods:   none

D, [2017-01-28T11:00:42.195489 #4407] DEBUG -- : Click methods:    none

D, [2017-01-28T11:00:42.195569 #4407] DEBUG -- : Disable-w-typing: n/a

D, [2017-01-28T11:00:42.195648 #4407] DEBUG -- : Accel profiles:   n/a

D, [2017-01-28T11:00:42.195726 #4407] DEBUG -- : 

[... 148 lines cut ...]
D, [2017-01-28T11:00:42.202083 #4407] DEBUG -- : Tap-to-click:     n/a

D, [2017-01-28T11:00:42.202179 #4407] DEBUG -- : Tap-and-drag:     n/a

D, [2017-01-28T11:00:42.202253 #4407] DEBUG -- : Tap drag lock:    n/a

D, [2017-01-28T11:00:42.202320 #4407] DEBUG -- : Left-handed:      n/a

D, [2017-01-28T11:00:42.202387 #4407] DEBUG -- : Nat.scrolling:    n/a

D, [2017-01-28T11:00:42.202896 #4407] DEBUG -- : Middle emulation: n/a

D, [2017-01-28T11:00:42.202962 #4407] DEBUG -- : Calibration:      n/a

D, [2017-01-28T11:00:42.203022 #4407] DEBUG -- : Scroll methods:   none

D, [2017-01-28T11:00:42.203109 #4407] DEBUG -- : Click methods:    none

D, [2017-01-28T11:00:42.203178 #4407] DEBUG -- : Disable-w-typing: n/a

D, [2017-01-28T11:00:42.203259 #4407] DEBUG -- : Accel profiles:   n/a

D, [2017-01-28T11:00:42.203323 #4407] DEBUG -- : 

D, [2017-01-28T11:00:42.203487 #4407] DEBUG -- : {:libinput_command=>"stdbuf -oL -- libinput-debug-events --enable-dwt --device /dev/input/event5"}

Saboteur777, 2017-01-28:
Hi, what `libinput-tools` version do you have? This might be related to https://github.com/iberianpig/fusuma/issues/26

crpig, 2017-01-28:
The libinput-tools version is 1.2.3-1ubuntu1

Saboteur777, 2017-01-30:
Did you added your user to input group? If yes, did you log out and lag back in?

crpig, 2017-02-01:
Here is my input group, zdw is current user.
```
~  cat /etc/group | grep input
input:x:106:zdw
```

And I even reboot my notebook, but still the same behaviour.

iberianpig, 2017-02-13:
@crpig  Do you still have a problem? What are your environment(distribution/version) and Fusuma version?

iberianpig, 2017-02-25:
Please reopen if you still have a problem. Thank you.

closed 2017-02-25 by iberianpig

Ante994, 2019-06-04:
fusuma only working with sudo, whene call without sudu i got this:
`Traceback (most recent call last):
	2: from /usr/bin/fusuma:23:in `<main>'
	1: from /usr/lib/ruby/2.5.0/rubygems.rb:303:in `activate_bin_path'
/usr/lib/ruby/2.5.0/rubygems.rb:284:in `find_spec_for_exe': can't find gem fusuma (>= 0.a) with executable fusuma (Gem::GemNotFoundException)
`
i installed fusuma following instructions, but still not working

iberianpig, 2019-06-05:
Have you already added $USER to input group then restart session?

Ante994, 2019-06-05:
yes, `cat /etc/group | grep input` gives `input:x:104:ante`

Ante994, 2019-06-05:
maybe have something with ruby ? ruby -v gives:
<pre>ruby 2.6.3p62 (2019-04-16 revision 67580) [x86_64-linux]</pre>

but in error message ruby 2.5.0 is mentioned

iberianpig, 2019-06-05:
It looks different from the runtime version. Try reinstalling fusuma. And if you use rbenv, try reinstalling without sudo.

Ante994, 2019-06-05:
after uninstall, and installing folowing instruction i got: `sudo: fusuma: command not found`
also made restart, still not work

it is not installed under /usr/bin

Ante994, 2019-06-05:
ok, let's say i solved it! i found out install location, it is under `~/.gem/gems/fusuma-0.11.1/exe/` from there fusuma command is executable, now i just setup to run with after boot

@iberianpig tnx for giving feedback

iberianpig, 2019-06-05:
I'm glad it worked!😄