iberianpig/fusuma #29: parse_finger_directions: undefined method `tr' for nil:NilClass (NoMethodError)

peyloride, 2017-02-02:
Hello,

I'm trying to use fusuma on my Archlinux setup. I'm using latest packages and linux-zen kernel. Libinput-debug-events shows the gestures but fusuma throws error. My ruby version is 2.4.0

fusuma -v:
```
D, [2017-02-02T11:45:42.287273 #2946] DEBUG -- : Device:           Power Button

D, [2017-02-02T11:45:42.288038 #2946] DEBUG -- : Kernel:           /dev/input/event3

D, [2017-02-02T11:45:42.288700 #2946] DEBUG -- : Group:            1

D, [2017-02-02T11:45:42.289257 #2946] DEBUG -- : Seat:             seat0, default

D, [2017-02-02T11:45:42.289870 #2946] DEBUG -- : Capabilities:     keyboard 

D, [2017-02-02T11:45:42.290368 #2946] DEBUG -- : Tap-to-click:     n/a

D, [2017-02-02T11:45:42.290874 #2946] DEBUG -- : Tap-and-drag:     n/a

D, [2017-02-02T11:45:42.291352 #2946] DEBUG -- : Tap drag lock:    n/a

D, [2017-02-02T11:45:42.291838 #2946] DEBUG -- : Left-handed:      n/a

D, [2017-02-02T11:45:42.292327 #2946] DEBUG -- : Nat.scrolling:    n/a

D, [2017-02-02T11:45:42.293041 #2946] DEBUG -- : Middle emulation: n/a

D, [2017-02-02T11:45:42.293600 #2946] DEBUG -- : Calibration:      n/a

D, [2017-02-02T11:45:42.294095 #2946] DEBUG -- : Scroll methods:   none

D, [2017-02-02T11:45:42.294563 #2946] DEBUG -- : Click methods:    none

D, [2017-02-02T11:45:42.295036 #2946] DEBUG -- : Disable-w-typing: n/a

D, [2017-02-02T11:45:42.295509 #2946] DEBUG -- : Accel profiles:   n/a

D, [2017-02-02T11:45:42.295982 #2946] DEBUG -- : Rotation:         n/a

D, [2017-02-02T11:45:42.296519 #2946] DEBUG -- : 

D, [2017-02-02T11:45:42.297021 #2946] DEBUG -- : Device:           Video Bus

D, [2017-02-02T11:45:42.297506 #2946] DEBUG -- : Kernel:           /dev/input/event6

D, [2017-02-02T11:45:42.300149 #2946] DEBUG -- : Group:            2

D, [2017-02-02T11:45:42.300334 #2946] DEBUG -- : Seat:             seat0, default

D, [2017-02-02T11:45:42.300437 #2946] DEBUG -- : Capabilities:     keyboard 

D, [2017-02-02T11:45:42.300524 #2946] DEBUG -- : Tap-to-click:     n/a

D, [2017-02-02T11:45:42.300614 #2946] DEBUG -- : Tap-and-drag:     n/a

D, [2017-02-02T11:45:42.300694 #2946] DEBUG -- : Tap drag lock:    n/a

D, [2017-02-02T11:45:42.306339 #2946] DEBUG -- : Left-handed:      n/a

D, [2017-02-02T11:45:42.306855 #2946] DEBUG -- : Nat.scrolling:    n/a

D, [2017-02-02T11:45:42.307185 #2946] DEBUG -- : Middle emulation: n/a

D, [2017-02-02T11:45:42.307295 #2946] DEBUG -- : Calibration:      n/a

D, [2017-02-02T11:45:42.307384 #2946] DEBUG -- : Scroll methods:   none

D, [2017-02-02T11:45:42.307469 #2946] DEBUG -- : Click methods:    none

[... 186 lines cut ...]
D, [2017-02-02T11:45:42.316759 #2946] DEBUG -- : Rotation:         n/a

D, [2017-02-02T11:45:42.316829 #2946] DEBUG -- : 

D, [2017-02-02T11:45:42.316975 #2946] DEBUG -- : {:libinput_command=>"stdbuf -oL -- libinput-debug-events --enable-dwt --device /dev/input/event5"}
```

when i do 3 finger gestures it throws the following error message;

```
/home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/lib/fusuma/gesture_action.rb:46:in `parse_finger_directions': undefined method `tr' for nil:NilClass (NoMethodError)
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/lib/fusuma/gesture_action.rb:33:in `gesture_action_arguments'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/lib/fusuma/gesture_action.rb:22:in `initialize_by'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:38:in `block (2 levels) in read_libinput'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:37:in `each'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:37:in `block in read_libinput'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/2.4.0/open3.rb:205:in `popen_run'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/2.4.0/open3.rb:95:in `popen3'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:36:in `read_libinput'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/lib/fusuma.rb:21:in `run'
	from /home/peyloride/.rbenv/versions/2.4.0/lib/ruby/gems/2.4.0/gems/fusuma-0.2.6/exe/fusuma:21:in `<top (required)>'
	from /home/peyloride/.rbenv/versions/2.4.0/bin/fusuma:22:in `load'
	from /home/peyloride/.rbenv/versions/2.4.0/bin/fusuma:22:in `<main>'

```

tobac, 2017-02-02:
This has already been reported as issue #26

peyloride, 2017-02-03:
Sorry, you are right. I saw that post but i just think similiar issue, not the same.

iberianpig, 2017-02-05:
@peyloride please update fusuma, and please confirm fix your problem.

iberianpig, 2017-02-13:
If you still have problems, please reopen this ticket.

closed 2017-02-13 by iberianpig