iberianpig/fusuma #31: trouble detecting Left and Right swipes

marcosfede, 2017-02-20:
First of all thanks for fusuma, it's really useful.

 atm the up and down swipes detect perfectly (with 3 and 4 fingers) but the left and right swipes are really difficult to trigger, I can trigger them maybe 1 out of 20 attempts.

 Is this a know issue or something I can solve from my end?

 I'm running it on a chromebook pixel running elementary OS loki.

iberianpig, 2017-02-20:
@marcosfede Have you tried to tweak threshold with editing `~/.config/fusuma/config.yml`? (see: https://github.com/iberianpig/fusuma#sample-default-keymap-for-elementary-os)

the0demiurge, 2017-02-21:
@iberianpig Aatually, it is hard to trigger all gestures even the threshold set to zero. I guess it is because that this software's orientation judging is strict, namely I must swipe very closely vertical or horizontal to trigger that.
I can trigger when 3 or 4 fingers swipe vertically 1 out of 5 attempts.
Ubuntu 16.04
Hope to improve this, thanks!

marcosfede, 2017-02-21:
@iberianpig tweaking threshold resulted in better detection, thanks!. also I realized I was aligning my fingers vertically to do a horizontal swipe to the left or right,   could that be the problem? If I now align my fingers horizontally when I swipe left or right it triggers much better. (I think this condition could be relaxed a little, since sometimes I don't have my fingers perfectly aligned and then the swipe doesn't trigger)

iberianpig, 2017-02-25:
It may be caused by libinput-debug-events cannot recognize gestures as swipe.
Please paste a output of `libinput-debug-events` after swiping on touchpad.

the0demiurge, 2017-02-25:
@iberianpig I just tried these swipes
`event2 	DEVICE_ADDED     Power Button                      seat0 default group1 cap:k
event4 	DEVICE_ADDED     Video Bus                         seat0 default group2 cap:k
event1 	DEVICE_ADDED     Power Button                      seat0 default group3 cap:k
event9 	DEVICE_ADDED     Lenovo EasyCamera                 seat0 default group4 cap:k
event6 	DEVICE_ADDED     RAPOO RAPOO 2.4G Wireless Device  seat0 default group5 cap:k
event7 	DEVICE_ADDED     RAPOO RAPOO 2.4G Wireless Device  seat0 default group6 cap:kp left scroll-nat scroll-button
event8 	DEVICE_ADDED     Ideapad extra buttons             seat0 default group7 cap:k
event3 	DEVICE_ADDED     AT Translated Set 2 keyboard      seat0 default group8 cap:k
event5 	DEVICE_ADDED     SynPS/2 Synaptics TouchPad        seat0 default group9 cap:pg	size 104.43/67.50mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on
event5 	POINTER_MOTION    +5.12s	  0.00/ -0.59
event5 	POINTER_MOTION    +5.16s	  0.00/ -0.85
event5 	POINTER_MOTION    +5.17s	  0.00/ -0.19
event5 	POINTER_MOTION    +5.19s	  0.00/ -1.58
event5 	POINTER_MOTION    +5.20s	  0.00/ -3.70
event5 	POINTER_MOTION    +5.21s	  0.00/ -4.01
event5 	POINTER_MOTION    +5.22s	  0.00/ -6.06
event5 	POINTER_MOTION    +5.23s	  0.00/ -8.99
event5 	POINTER_MOTION    +5.24s	 -0.11/-10.51
event5 	POINTER_MOTION    +5.25s	 -0.38/-15.12
event5 	POINTER_MOTION    +5.27s	 -0.37/-21.07
event5 	POINTER_MOTION    +5.29s	 -0.13/-23.32
event5 	POINTER_MOTION    +5.30s	  0.00/-16.26
event5 	POINTER_MOTION    +5.31s	  0.00/ -7.02
event5 	POINTER_MOTION    +5.32s	  0.00/ -3.02
event5 	POINTER_MOTION    +5.35s	  0.00/  1.68
event5 	POINTER_MOTION    +5.36s	 -0.18/  3.49
event5 	POINTER_MOTION    +5.80s	  4.50/  0.76
event5 	POINTER_MOTION    +5.81s	  2.16/  2.20
event5 	POINTER_MOTION    +5.83s	  0.28/  3.63
event5 	POINTER_MOTION    +5.83s	  0.28/  3.82
event5 	POINTER_MOTION    +5.84s	  0.10/  4.28
event5 	POINTER_MOTION    +5.85s	  0.10/  4.48
event5 	POINTER_MOTION    +5.87s	  0.30/  6.46
event5 	POINTER_MOTION    +5.88s	  0.42/  8.76
event5 	POINTER_MOTION    +5.88s	  0.44/  7.29
event5 	POINTER_MOTION    +5.89s	  0.44/  5.99
event5 	POINTER_MOTION    +5.91s	  2.42/  8.07
event5 	POINTER_MOTION    +5.92s	  4.30/  6.95
event5 	POINTER_MOTION    +5.93s	  2.15/  4.69
event5 	POINTER_MOTION    +7.29s	 -4.40/  1.85
event5 	POINTER_MOTION    +7.33s	 -0.46/  1.57
event5 	POINTER_MOTION    +7.35s	  1.85/  0.00
event5 	POINTER_MOTION    +7.36s	  3.00/ -1.21
event5 	POINTER_MOTION    +7.37s	  5.29/ -2.44
event5 	POINTER_MOTION    +7.38s	  6.55/ -1.30
event5 	POINTER_MOTION    +7.39s	 13.02/  0.00
event5 	POINTER_MOTION    +7.40s	 26.66/  0.00
event5 	POINTER_MOTION    +7.41s	 25.01/ -0.29
event5 	POINTER_MOTION    +7.42s	 16.93/ -0.58
event5 	POINTER_MOTION    +7.43s	 15.91/ -0.71
event5 	POINTER_MOTION    +7.44s	 16.15/ -0.84
event5 	POINTER_MOTION    +7.81s	 -2.06/  0.00
event5 	POINTER_MOTION    +7.82s	 -1.75/  0.00
event5 	POINTER_MOTION    +7.83s	 -0.94/  0.00
event5 	POINTER_MOTION    +7.84s	 -0.84/  0.00
event5 	POINTER_MOTION    +7.85s	 -3.28/  0.08
event5 	POINTER_MOTION    +7.86s	 -5.16/  1.82
event5 	POINTER_MOTION    +7.87s	 -5.06/  3.41
event5 	POINTER_MOTION    +7.88s	 -5.25/  1.67
event5 	POINTER_MOTION    +7.89s	 -5.53/  0.00
event5 	POINTER_MOTION    +7.90s	 -8.06/  0.08
event5 	POINTER_MOTION    +7.91s	-11.00/  0.17
event5 	POINTER_MOTION    +7.93s	-12.03/  0.17
event5 	POINTER_MOTION    +7.94s	-13.12/  0.26
event5 	POINTER_MOTION    +7.95s	-11.86/  0.28
event5 	POINTER_MOTION    +7.96s	 -6.75/  0.17
event5 	POINTER_MOTION    +7.97s	 -2.26/  0.00
event5 	POINTER_MOTION    +8.97s	  0.00/ -0.14
event5 	POINTER_AXIS      +9.08s	vert -13.17* horiz -2.06*
[... 190 lines cut ...]
    right: 
      shortcut: 'ctrl+shift+tab'
    up: 
      shortcut: 'ctrl+super+Up'
    down: 
      shortcut: 'ctrl+super+Down'
  4:
    left: 
      shortcut: 'alt+tab'
    right: 
      shortcut: 'alt+shift+tab'
    up: 
      shortcut: 'ctrl+alt+Down'
    down: 
      shortcut: 'ctrl+alt+Up'
pinch:
  in:
    shortcut: 'ctrl+plus'
  out:
     shortcut: 'ctrl+minus'

threshold:
  swipe: 0
  pinch: 0.5
`

LitterChen, 2017-03-09:
Firstly thanks for fusuma.
But it is hard to trigger even the threshold set to small such as 0.1. I find the triggers will be better if I swipe 3 or 4 fingers after swipe 2 fingers.if I want continuous swipe to touch off a sequence shortcut.The triggers will be not good. 
Hope to improve this, thanks!

the0demiurge, 2017-04-08:
I found a solution to this problem by accidently run fusuma with root privileges, if not with root privileges, the fusuma may not work reliably with many consequences. However, it is also not very sensitive when swiping left/right.

iberianpig, 2017-04-22:
@LitterChen The option is added for tweaking an interval until next a swipe/pinch with ver0.3.0.
Please try with updating fusuma.

MisterPresident, 2017-05-08:
I am having problems detecting 5-finger actions (if I run `fusuma -v` it tells me that i just use 4 fingers but doesn't perform 4-finger action too.
`libinput-debug-events` doesn't show anything when i preform a 5 finger swipe but with your old pear script xSwipe my 5 finger swipe works.
Do you have a tip for me?

iberianpig, 2017-05-08:
@MisterPresident Thank you for using my tools.
It indicates libinput drivers' detect up to 4 fingers. This issue may be libinput driver's issue.

closed 2017-07-14 by iberianpig