iberianpig/fusuma #32: Typed in fusuma after installation, but no response

naco-siren, 2017-02-27:


iberianpig, 2017-02-28:
@naco-siren  Please tell me the detail of your issue. (Distribution, Desktop Environment, kernel version)

naco-siren, 2017-02-28:
Ubuntu 16.04 LTS
Unity (default)
4.4 (default)

iberianpig, 2017-03-01:
Please paste a output of `libinput-debug-events` after swiping on touchpad.

naco-siren, 2017-03-01:
event8 	GESTURE_SWIPE_BEGIN  +1.41s	3
event8 	GESTURE_SWIPE_UPDATE  +1.41s	3 -0.46/-6.96 (-1.31/-19.69 unaccelerated)
event8 	GESTURE_SWIPE_UPDATE  +1.42s	3 -1.70/-26.37 (-3.12/-48.39 unaccelerated)
event8 	GESTURE_SWIPE_UPDATE  +1.44s	3 -2.53/-47.25 (-3.28/-61.35 unaccelerated)
event8 	GESTURE_SWIPE_END  +1.47s	3

event8 	GESTURE_SWIPE_BEGIN  +5.36s	4
event8 	GESTURE_SWIPE_UPDATE  +5.36s	4 -3.91/-25.05 (-6.77/-43.31 unaccelerated)
event8 	GESTURE_SWIPE_END  +5.38s	4

event8 	GESTURE_SWIPE_BEGIN  +7.32s	4
event8 	GESTURE_SWIPE_UPDATE  +7.32s	4 -13.04/ 2.61 (-32.60/ 6.52 unaccelerated)
event8 	GESTURE_SWIPE_UPDATE  +7.34s	4 -74.79/14.53 (-101.99/19.81 unaccelerated)
event8 	GESTURE_SWIPE_UPDATE  +7.36s	4 -113.29/21.85 (-141.61/27.31 unaccelerated)
event8 	GESTURE_SWIPE_UPDATE  +7.37s	4 -71.56/13.58 (-89.44/16.98 unaccelerated)
event8 	GESTURE_SWIPE_UPDATE  +7.39s	4 -27.90/ 5.39 (-36.29/ 7.01 unaccelerated)
event8 	GESTURE_SWIPE_UPDATE  +7.40s	4 -4.69/ 1.68 (-8.24/ 2.95 unaccelerated)
event8 	GESTURE_SWIPE_END  +7.42s	4
event8 	GESTURE_PINCH_BEGIN  +9.95s	4
event8 	GESTURE_PINCH_UPDATE  +9.95s	4  0.38/ 0.00 ( 0.98/ 0.00 unaccelerated)  1.00 @ -0.02
event8 	GESTURE_PINCH_UPDATE  +9.96s	4  0.00/-0.18 ( 0.00/-0.98 unaccelerated)  1.00 @ -0.02
event8 	GESTURE_PINCH_UPDATE  +9.96s	4  0.27/ 0.27 ( 0.98/ 0.98 unaccelerated)  1.00 @ -0.04
event8 	GESTURE_PINCH_UPDATE  +9.97s	4  1.56/ 0.39 ( 3.94/ 0.98 unaccelerated)  0.99 @ -0.06
event8 	GESTURE_PINCH_END  +9.99s	4

naco-siren, 2017-03-01:
(Sequence: 3-finger, 4-finger and 4-finger)
Thanks!

ahmadmuzakki29, 2017-07-17:
I did a restart and still nothing happened when I type fusuma
my output of `libinput-debug-events` is pretty much the same as @naco-siren 
and no fusuma folder created in ~/.config

ubuntu gnome 16.04
kernel 4.8.0-58-generic

closed 2018-04-22 by stale[bot]