iberianpig/fusuma #34: External + Internal Trackpad?

christophersw, 2017-04-04:
I have a MacBook Pro with an internal multi-touch track pad which works perfectly. Trouble is, I can't get multi-touch to work on my external Apply Magic Trackpad connected via bluetooth. I realize this is likely not strictly a Fusuma issue, but I was really hoping you could point me in the right direction?

iberianpig, 2017-04-05:
Could you try stop and restart `fusuma`?
`fusuma` can read multi devices` touching, but the it can detect available devices only at starting `fusuma`.

If you have a problem, please paste output of `libinput-list-devices `

christophersw, 2017-04-05:
Still no luck, here's the relevant bits of my `libinput-list-devices` output:

> ...
> 
> Device:           Christopher’s Trackpad
> Kernel:           /dev/input/event8
> Group:            7
> Seat:             seat0, default
> Capabilities:     pointer 
> Tap-to-click:     n/a
> Tap-and-drag:     n/a
> Tap drag lock:    n/a
> Left-handed:      disabled
> Nat.scrolling:    disabled
> Middle emulation: n/a
> Calibration:      n/a
> Scroll methods:   button
> Click methods:    none
> Disable-w-typing: n/a
> Accel profiles:   flat*adaptive
> 
> Device:           Apple Inc. Apple Internal Keyboard / Trackpad
> Kernel:           /dev/input/event6
> Group:            8
> Seat:             seat0, default
> Capabilities:     keyboard 
> Tap-to-click:     n/a
> Tap-and-drag:     n/a
> Tap drag lock:    n/a
> Left-handed:      n/a
> Nat.scrolling:    n/a
> Middle emulation: n/a
> Calibration:      n/a
> Scroll methods:   none
> Click methods:    none
> Disable-w-typing: n/a
> Accel profiles:   n/a
> 
> Device:           bcm5974
> Kernel:           /dev/input/event9
> Group:            9
> Seat:             seat0, default
> Size:             69.17x50.22mm
> Capabilities:     pointer 
> Tap-to-click:     disabled
> Tap-and-drag:     enabled
> Tap drag lock:    disabled
> Left-handed:      disabled
> Nat.scrolling:    disabled
> Middle emulation: n/a
> Calibration:      n/a
> Scroll methods:   *two-finger edge 
> Click methods:    button-areas *clickfinger 
> Disable-w-typing: enabled
> Accel profiles:   none
>

iberianpig, 2017-04-22:
@christophersw Please update fusuma and restart fusuma.

closed 2017-04-22 by iberianpig