iberianpig/fusuma #35: How about adding adjustable interval threshold?

joshuatshaffer, 2017-04-20:
I would like the option to make gestures in rapid succession. I am using Fusuma for switching workspaces in a similar fashion to macOS. I really like Fusuma because it gives me a very quick way to move between several desktops, but it is annoying to have to wait a half second between swipes.

2edff206, 2017-04-21:
I installed fusuma today because Touchegg doesn't work anymore on 17.04.
I was about to make the same request, I am glad someone had the same idea ! It's annoying to have to wait to switch multiple workspaces

Thanks for your work

2edff206, 2017-04-21:
Ok I solved it myself. For info you just have to edit this file /var/lib/gems/2.3.0/gems/fusuma-0.2.7/lib/fusuma/swipe.rb
Near the beginning of the file change INTERVAL=0.5 to INTERVAL=0.1

joshuatshaffer, 2017-04-21:
Ah, great! Yeah, I'm also using this because Ubuntu 17.04 killed Touchégg. Thanks for the work around. Still, I think this is a good candidate for a new config option.

joshuatshaffer, 2017-04-21:
I went ahead and added this feature my self. The changes are in pull request #36. I hope I did everything alright. This is my first time using ruby and my first time contributing to a real project.

iberianpig, 2017-04-22:
@joshuatshaffer Thanks a lot for your PR!

closed 2017-04-22 by iberianpig