iberianpig/fusuma #37: Swipes not detected in xfce

rubenwardy, 2017-04-26:
I can't seem to find a comprehensive setup guide which specifies the supported drivers etc

```
rubenwardy:~ $ xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ETPS/2 Elantech Touchpad                	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Asus Wireless Radio Control             	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ USB Camera                              	id=10	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=11	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=12	[slave  keyboard (3)]
```

libinput-debug-events

```
event5 	POINTER_MOTION    +1.25s	  0.10/  0.47
event5 	POINTER_MOTION    +1.26s	  0.20/  0.66
event5 	POINTER_MOTION    +1.27s	  0.10/  0.78
event5 	POINTER_MOTION    +1.29s	  0.00/  0.80
event5 	POINTER_MOTION    +1.30s	  0.00/  0.92
event5 	POINTER_MOTION    +1.31s	  0.00/  1.36
event5 	POINTER_MOTION    +1.32s	  0.23/  2.22
event5 	POINTER_MOTION    +1.34s	  0.75/  3.41
event5 	POINTER_MOTION    +1.35s	  1.14/  4.18
event5 	POINTER_MOTION    +1.36s	  1.27/  4.31
event5 	POINTER_MOTION    +1.38s	  1.14/  3.81
event5 	POINTER_MOTION    +1.39s	  1.02/  3.08
event5 	POINTER_MOTION    +1.41s	  0.89/  2.34
event5 	POINTER_MOTION    +1.43s	  0.13/  0.37
event5 	POINTER_MOTION    +1.44s	  0.13/  0.00
event5 	POINTER_AXIS      +1.92s	vert -5.04* horiz 0.00
event5 	POINTER_AXIS      +1.94s	vert -5.91* horiz 0.00
event5 	POINTER_AXIS      +1.95s	vert -6.64* horiz 0.00
event5 	POINTER_AXIS      +1.96s	vert -6.58* horiz 0.00
event5 	POINTER_AXIS      +1.98s	vert -5.84* horiz 0.00
event5 	POINTER_AXIS      +1.99s	vert -4.68* horiz 0.00
event5 	POINTER_AXIS      +2.00s	vert -3.20* horiz -0.06*
event5 	POINTER_AXIS      +2.01s	vert -2.09* horiz -0.13*
event5 	POINTER_AXIS      +2.03s	vert -1.48* horiz -0.06*
event5 	POINTER_AXIS      +2.04s	vert -0.98* horiz 0.00
event5 	POINTER_AXIS      +2.36s	vert 0.00* horiz 0.00*
event5 	POINTER_MOTION    +2.82s	  0.57/  0.00
event5 	POINTER_MOTION    +2.83s	  1.55/  0.36
event5 	POINTER_MOTION    +2.84s	  4.44/  1.50
event5 	POINTER_MOTION    +2.86s	  4.95/  1.72
event5 	POINTER_MOTION    +2.87s	  4.06/  1.11
event5 	POINTER_MOTION    +2.88s	  4.06/  0.98
event5 	POINTER_MOTION    +2.89s	  4.19/  0.98
```

/usr/share/X11/xorg.list.d/20-natural-scrolling.conf

```
Section "InputClass"
        Identifier "Natural Scrolling"
	MatchIsTouchpad "on"
        MatchDevicePath "/dev/input/event*"
        Option "VertScrollDelta" "-75"
        Option "HorizScrollDelta" "75"
        Option "DialDelta" "-75"
EndSection
```

Looks like I use Synaptics:

```
rubenwardy:~ $ dmesg | grep -i 'syna'
[    1.602604] psmouse serio4: elantech: Synaptics capabilities query result 0x10, 0x14, 0x0e.
```

rubenwardy, 2017-04-26:
I'm an idiot, turns out the default you give does forward/back on two finger swipe :P

rubenwardy, 2017-04-26:
Sorry if this is the wrong place, there doesn't seem to be a forum. It would be really good to have a website or wiki with setup guides, example systemd services, and info on confirmed working distros

kate-lim, 2017-05-25:
I'm experiencing a similar issue. Originally I didn't specify any two finger swipes but fusuma seems to have messed up the directions of my scrolling since I installed it. I tried specifying two finger swipes like this:

`swipe:
    2:
        left:
            shortcut: 'alt+Left'
        right:
            shortcut: 'alt+Right'
        up:
            shortcut: 'Down'
        down:
            shortcut: 'Up'
`
Still doesn't fix the problem.

iberianpig, 2017-06-02:
Fusuma does not support two finger swipes.
It's by design.

bvarner, 2017-06-17:
Could you elaborate on why? I'm curious.... very very curious.

I miss having the universal 'back' for the two-finger swipes I got accustomed to in OS X.

iberianpig, 2017-06-20:
@bvarner it's driver's matter.  libinput driver doesn't recognize 2 fingers action as swipe.

closed 2017-07-14 by iberianpig