iberianpig/fusuma #38: 3 Finger Grab

fazakis, 2017-04-27:
Is it possible to have 3 fingers grab (e.g. grab a window and move it) gesture or is it possible to emulate it some way like (3 finger long press to grab)?

iberianpig, 2017-05-08:
libinput doesn't output  the information when staying in touchpad's same points. So that it is difficult to detect long-press.

closed 2017-05-08 by iberianpig

iberianpig, 2020-03-20:
@fazakis  I released [fusuma-plugin-tap](https://github.com/iberianpig/fusuma-plugin-tap). 
This plugin detects 1~4 fingers tap gestures and hold(long press) gestures.

install fusuma-plugin and fusuma-plugin-sendkey
```sh
sudo gem install fusuma-plugin-tap fusuma-plugin-sendkey
```

Then, modify your `~/.config/fusuma/config.yml` .
```yml
hold:
  3:
    sendkey: "LEFTALT+F7" # drag current window

plugin:
  inputs:
    libinput_command_input:
      enable-tap: true
      verbose: true
```