iberianpig/fusuma #43: Can't boot fusuma in terminal

JessyTang, 2017-07-05:
Hello guys, I have follow the README's introduction about fusuma to download and install. But I can't boot it. There is the information about debug mode.
```
$ fusuma -v
```
[image]
Waiting for your reply!

liuliu7616, 2017-07-09:
I have the same question

ivanovmi, 2017-08-26:
@JessyTang @liuliu7616 try to run it with sudo privileges. Hope this helps :smile:

liuliu7616, 2017-08-27:
thank you  @ivanovmi ,I can run the program now,but when i touch the touchpad whit three fingers , it print some error codes

yangyang@yangyang-X550JX:~$ sudo fusuma
/usr/lib/ruby/2.3.0/psych.rb:377:in `parse': (/home/yangyang/.config/fusuma/config.yml): found character that cannot start any token while scanning for the next token at line 2 column 1 (Psych::SyntaxError)
	from /usr/lib/ruby/2.3.0/psych.rb:377:in `parse_stream'
	from /usr/lib/ruby/2.3.0/psych.rb:325:in `parse'
	from /usr/lib/ruby/2.3.0/psych.rb:252:in `load'
	from /usr/lib/ruby/2.3.0/psych.rb:471:in `block in load_file'
	from /usr/lib/ruby/2.3.0/psych.rb:471:in `open'
	from /usr/lib/ruby/2.3.0/psych.rb:471:in `load_file'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma/config.rb:36:in `reload'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma/config.rb:31:in `initialize'
	from /usr/lib/ruby/2.3.0/singleton.rb:142:in `new'
	from /usr/lib/ruby/2.3.0/singleton.rb:142:in `block in instance'
	from /usr/lib/ruby/2.3.0/singleton.rb:140:in `synchronize'
	from /usr/lib/ruby/2.3.0/singleton.rb:140:in `instance'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma/config.rb:14:in `threshold'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma/swipe.rb:40:in `threshold'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma/swipe.rb:26:in `enough_distance?'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma/swipe.rb:20:in `enough?'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma/action_stack.rb:31:in `detect_direction'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma/action_stack.rb:13:in `gesture_info'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:42:in `block (2 levels) in read_libinput'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:37:in `each'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:37:in `block in read_libinput'
	from /usr/lib/ruby/2.3.0/open3.rb:205:in `popen_run'
	from /usr/lib/ruby/2.3.0/open3.rb:95:in `popen3'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:36:in `read_libinput'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:21:in `run'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/exe/fusuma:21:in `<top (required)>'
	from /usr/local/bin/fusuma:23:in `load'
	from /usr/local/bin/fusuma:23:in `<main>'

ivanovmi, 2017-08-27:
@liuliu7616 looks like your file indented by tabs. Try to change each tab character to double spaces

liuliu7616, 2017-08-27:
@ivanovmi thank you very much ,it can run now ,I am so careless

closed 2018-03-01 by JessyTang