iberianpig/fusuma #46: How to autostart fusuma in the background on login?

Rodancoci, 2017-07-21:
I'm running Unity on Ubuntu 17.04.

So far I've been manually running
`$ fusuma & disown`
And then closing the terminal, which works well enough.

However, I wish I could get fusuma to run by itself whenever I login. I've tried adding the command 'fusuma'  to Ubuntu's "Startup Applications" without success. Upon rebooting and logging in neither fusuma nor ruby are running.

Any suggestions?

iberianpig, 2017-07-22:
@Rodancoci 

You can find  correct path with below command.
```
$ which fusuma
```

then please register the path you found and try restart.

Rodancoci, 2017-07-23:
I tried setting the full path output by
`$ which fusuma`
As the command on the startup item but I got the same result.

Launching the full path manually from the terminal after logging in works, however.

iberianpig, 2017-07-27:
Please record log and check it on your environment.
`path/to/your/fusuma -v > /tmp/fusuma.log`
restart your machine, then `cat /tmp/fsuma.log`

Rodancoci, 2017-08-02:
Here are the contents of the log: https://pastebin.com/MNcQ7Mgp

Rodancoci, 2017-08-02:
I managed to get it working how I wanted. What finally worked for me was adding the following line at the end of `~/.profile`:
`/home/usr/.rvm/gems/ruby-2.4.0/wrappers/fusuma & disown`

kontrollanten, 2017-08-16:
Thanks @Rodancoci for the help. I got it working with the following command.
`printf "$(which fusuma) & disown \n" >> .profile`

But .profile doesn't feel like the natural place for this command, even if it works. I'm pretty new into Linux, but wouldn't it be better if the application created a script in init.d? Then it would be more of a plug n play application.

iberianpig, 2017-08-20:
@kontrollanten Oh, that's a nice idea. Could you make the pull-request?

kontrollanten, 2017-08-20:
I'm new into both ruby and Linux. But I'll give it some tries.

worldofpeace, 2017-11-08:
What about people who use zsh etc.?

rmasclef, 2018-02-26:
@Rodancoci in a more user friendly way, you can use `startup application preferences` to start any software you want at user login.

For example, I added `firefox`, `slack` and `fusuma` in order to start those three apps when I login.

ouzc12358, 2018-04-30:
@rmasclef  thanks! simple and useful ! it takes my 2 days

closed 2018-07-06 by stale[bot]

JafarBadour, 2020-04-14:
I am using ubuntu 19
When I add fusuma to startup applications It doesnt work. However when I run 
```
sudo fusuma &> dev/null
```

It works perfectly. 

I tried this one 
[image]

and this one:
[image]

Neither works, i think because it doesnt accept sudo but I dont know how to fix!

iberianpig, 2020-04-14:
Please try adding $USER to input group and configure autostart without sudo.

JafarBadour, 2020-04-14:
I am not sure how is that possible? how can I add $USER ?

iberianpig, 2020-04-14:
Please read README.

https://github.com/iberianpig/fusuma/blob/master/README.md#autostart-gnome-session-properties

JafarBadour, 2020-04-14:
Yep this worked for me

## Autostart (gnome-session-properties)
Check the path where you installed fusuma with $ which fusuma
Open ```$ gnome-session-properties```
Add ```Fusuma``` and enter the location where the above path was checked in the command input field
Add the ```-d``` option at the end of the command input field

## Screenshot of the configuration on my device

[image]

athulkrishna2015, 2021-06-25:
> Yep this worked for me, thanks!!
> ## Autostart (gnome-session-properties)
> 
> Check the path where you installed fusuma with $ which fusuma
> Open `$ gnome-session-properties`
> Add `Fusuma` and enter the location where the above path was checked in the command input field
> Add the `-d` option at the end of the command input field
> ## Screenshot of the configuration on my device
> 
> [image]

what is the meaning of `-d`

JafarBadour, 2021-06-25:
I think -d is for detached mode @athulkrishna2015