iberianpig/fusuma #48: Problem when I run fusuma on the bash

yoyozaza, 2017-10-01:
When I run the fusuma on the bash nothing appears
And when I force the interruption with ^C :
`^C/var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:37:in `each': Interrupt
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:37:in `block in read_libinput'
	from /usr/lib/ruby/2.3.0/open3.rb:205:in `popen_run'
	from /usr/lib/ruby/2.3.0/open3.rb:95:in `popen3'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:36:in `read_libinput'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/lib/fusuma.rb:21:in `run'
	from /var/lib/gems/2.3.0/gems/fusuma-0.3.3/exe/fusuma:21:in `<top (required)>'
	from /usr/local/bin/fusuma:23:in `load'
	from /usr/local/bin/fusuma:23:in `<main>'
`

yoyozaza, 2017-10-01:
Maybe this is because i'm not on the input group?

worldofpeace, 2017-10-03:
When you run `fusima` in bash it shouldn't display anything. The only way for to tell if it's running correctly is by attempting to use a gesture.

 The README shows you how to add yourself to the input group. Try doing that and seeing if it works. Make sure you log out and then log back in. What DE are you using?

iberianpig, 2017-10-03:
Did you logout and login your session after assign input group?
https://github.com/iberianpig/fusuma/issues/33

yoyozaza, 2017-10-03:
I readed the README but I didn't understand, for being on the input group I need to write this :
`sudo gpasswd -a $USER input`
what i need to write instead of $USER ?
and then I need to  reboot the system ?
I am on ubuntu 16.04 
Sorry for all these question, I am a beginner on linux ;)

yoyozaza, 2017-10-03:
It's ok i copied the code and it's work fine

closed 2017-10-03 by yoyozaza