iberianpig/fusuma #50: assigning commands

tristan-k, 2017-10-22:
Is is possible to assign arbitrary commands to a specific gesture? For example [skippy-xd](https://github.com/hotice/webupd8/blob/master/skippy-xd-fix) to recreate a [mission-control](https://support.apple.com/de-de/HT204100) like overview.

```
swipe:
  4: 
    up: 
      shortcut: '/usr/local/bin/skippy-xd-fix' 
```

iberianpig, 2017-10-24:
@tristan-k You can only use **BINDABLE KEY COMBINATIONS** on fusuma's config.yml.
Fusuma triggers shortcuts configured by user.
So you should assign a shortcut to your command, before using fusuma. 
(I recommend you to use xbindkeys.)

yellowgh0st, 2017-12-11:
Would be great option to assign arbitrary commands too.
It could be super handy even on wayland.

iberianpig, 2018-01-05:
OK, I'll implement assign any command instead of xdotool key events.

iberianpig, 2018-03-12:
@augustin-b I implemented assigning commands version. https://github.com/iberianpig/fusuma/tree/feature/exec_command
Please try this branch.

iberianpig, 2018-04-15:
@augustin-b @tristan-k Released with new version. Please update fusuma.

closed 2018-04-15 by iberianpig