iberianpig/fusuma #51: Shortcut doesn't work

yoyozaza, 2017-10-27:
Hi, 

Thank you for this program

But I have a problem when I wrote `up:
      shortcut: 'super+shift+w'
`
for the 3 finger it's not work

iberianpig, 2017-11-01:
@yoyozaza other events are works so far?
Please check  `fusuma -v` and paste the logs with 3finger swipes?
And paste your config.yml.

yoyozaza, 2017-11-01:
`D, [2017-11-01T18:58:11.506409 #31278] DEBUG -- : {:libinput_command=>"stdbuf -oL -- libinput-debug-events --enable-dwt --device /dev/input/event5"}
D, [2017-11-01T18:58:12.976678 #31278] DEBUG -- : {:time=>"+1.47s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.31", :y=>"-21.05"}, :zoom=>nil}}
D, [2017-11-01T18:58:13.015594 #31278] DEBUG -- : {:time=>"+1.51s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"5.55", :y=>"-36.16"}, :zoom=>nil}}
D, [2017-11-01T18:58:13.066904 #31278] DEBUG -- : {:time=>"+1.56s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"5.38", :y=>"-21.31"}, :zoom=>nil}}
D, [2017-11-01T18:58:13.067131 #31278] DEBUG -- : {:x=>3.746666666666666, :y=>-26.173333333333332}
D, [2017-11-01T18:58:13.068971 #31278] DEBUG -- : trigger keyevent: super+shift+w
D, [2017-11-01T18:58:13.140525 #31278] DEBUG -- : {:time=>"+1.58s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
`
and my config.yml : 
`swipe:

  3:
    left:
      shortcut: 'ctrl+alt+Right'
    right:
      shortcut: 'ctrl+alt+Left'
    up:
      shortcut: 'super+shift+w'
    down:
      shortcut: 'ctrl+alt+Up'
  4:
    left:
      shortcut: 'alt+Left'
    right:
      shortcut: 'alt+Right'
    up:
      shortcut: 'super+shift+w'
    down:
      shortcut: 'super+w'
pinch:
  in:
    shortcut: ''
  out:
     shortcut: ''

threshold:
  swipe: 0.1
  pinch: 0.1

interval:
  swipe: 1.0
  pinch: 0.1

`

yoyozaza, 2017-11-01:
Sorry but the display of code doesn't work

yoyozaza, 2017-11-01:
But the other events like switch desktop work very well and thank you for this :+1:

Raensul, 2017-11-11:
hi!
for 4 fingers it works well, but with 3 finger smthng wrong...

Raensul, 2017-11-11:
libinput-list
```
Device:           SynPS/2 Synaptics TouchPad
Kernel:           /dev/input/event7
Group:            7
Seat:             seat0, default
Size:             106.73x63.39mm
Capabilities:     pointer 
Tap-to-click:     disabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *two-finger edge 
Click methods:    *button-areas clickfinger 
Disable-w-typing: enabled
Accel profiles:   none
Rotation:         n/a
```

fusuma -v
```
D, [2017-11-11T16:04:29.557390 #7432] DEBUG -- : Device:           Power Button

D, [2017-11-11T16:04:29.557617 #7432] DEBUG -- : Kernel:           /dev/input/event2

D, [2017-11-11T16:04:29.557746 #7432] DEBUG -- : Group:            1

D, [2017-11-11T16:04:29.557808 #7432] DEBUG -- : Seat:             seat0, default

D, [2017-11-11T16:04:29.557858 #7432] DEBUG -- : Capabilities:     keyboard 

D, [2017-11-11T16:04:29.557903 #7432] DEBUG -- : Tap-to-click:     n/a

D, [2017-11-11T16:04:29.557952 #7432] DEBUG -- : Tap-and-drag:     n/a

D, [2017-11-11T16:04:29.557997 #7432] DEBUG -- : Tap drag lock:    n/a

D, [2017-11-11T16:04:29.558042 #7432] DEBUG -- : Left-handed:      n/a

D, [2017-11-11T16:04:29.558086 #7432] DEBUG -- : Nat.scrolling:    n/a

D, [2017-11-11T16:04:29.558143 #7432] DEBUG -- : Middle emulation: n/a

D, [2017-11-11T16:04:29.558189 #7432] DEBUG -- : Calibration:      n/a

D, [2017-11-11T16:04:29.558236 #7432] DEBUG -- : Scroll methods:   none

D, [2017-11-11T16:04:29.558278 #7432] DEBUG -- : Click methods:    none

D, [2017-11-11T16:04:29.558322 #7432] DEBUG -- : Disable-w-typing: n/a

D, [2017-11-11T16:04:29.558364 #7432] DEBUG -- : Accel profiles:   n/a

D, [2017-11-11T16:04:29.558406 #7432] DEBUG -- : Rotation:         n/a

D, [2017-11-11T16:04:29.558448 #7432] DEBUG -- : 

D, [2017-11-11T16:04:29.558784 #7432] DEBUG -- : Device:           Video Bus

D, [2017-11-11T16:04:29.558833 #7432] DEBUG -- : Kernel:           /dev/input/event4

D, [2017-11-11T16:04:29.558887 #7432] DEBUG -- : Group:            2

D, [2017-11-11T16:04:29.558932 #7432] DEBUG -- : Seat:             seat0, default

D, [2017-11-11T16:04:29.558974 #7432] DEBUG -- : Capabilities:     keyboard 

[... 293 lines cut ...]
    down: 
      shortcut: 'ctrl+Tab'
  4:
    left: 
      shortcut: 'super+Left'
    right: 
      shortcut: 'super+Right'
    up: 
      shortcut: 'ctrl+alt+Down'
    down: 
      shortcut: 'super+d'
pinch:
  in:
    shortcut: 'ctrl+plus'
  out:
     shortcut: 'ctrl+minus'

threshold:
  swipe: 0.1
  pinch: 0.1

interval:
  swipe: 1.0
  pinch: 0.1
```

rmasclef, 2018-03-25:
same for me, I tried to add `super+shift+w` to `3.up` & `3.down` events but nothing happens :-1:  (`super+s` works though).

@iberianpig any idea ?

rmasclef, 2018-04-08:
@iberianpig did you manage to reproduce the problem yet ?

iberianpig, 2018-04-10:
@rmasclef No, I couldn't reproduce.
My environment is elemnetary OS(based on Ubuntu16.04.1).
Does this issue relate on Ubuntu version?

rmasclef, 2018-04-17:
Ho I forgot to give you the Ubuntu version, I tried on 17.10 and 16.04 :)

rmasclef, 2018-04-28:
FYI: I'm gonna test that on 18.04LTS soon :+1:

closed 2018-07-04 by stale[bot]

rmasclef, 2018-07-11:
@iberianpig maybe we can add a "workaround" in the readme telling people that they can change the shortcut if they encounter the same issue as I do ?

iberianpig, 2018-07-13:
Ok, I'll update README for adding custom config.