iberianpig/fusuma #56: Fusuma not working on Ubuntu 17.10 with Elantech Touchpad

rdnckreaper, 2018-01-05:
I have an Asus X75A and cannot get Fusuma to work. I have tried everything I can find online and cannot get the software to see my Elantech Touchpad.
Below are my terminal commands and responses to show what it happening. I installed everything in order, the way it needed to be and it wasn't until the end that I got the errors.

fusuma
E, [2018-01-05T02:25:05.905671 #5704] ERROR -- : Touchpad is not found

fusuma -v
D, [2018-01-05T02:22:53.544101 #5671] DEBUG -- : {:device_names=>[]}
E, [2018-01-05T02:22:53.544234 #5671] ERROR -- : Touchpad is not found

xinput
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ETPS/2 Elantech Touchpad                  id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Asus Wireless Radio Control               id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Sleep Button                              id=9    [slave  keyboard (3)]
    ↳ USB2.0 HD UVC WebCam: USB2.0 HD           id=10   [slave  keyboard (3)]
    ↳ Asus WMI hotkeys                          id=11   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=12   [slave  keyboard (3)]

Any help with this issue would be greatly appreciated as this is the first piece of software that can give me the gestures that I am looking for. I have tried touchegg and it wouldn't work and I read online that probably has something to do with Wayland but I couldn't figure it out. Then i came upon this and though I had found a true GEM (play on words...lol) and I still feel that it will be if I could just get it to work.

Thank you for your time.

iberianpig, 2018-01-05:
Please paste the output of 3 fingers swipes with running `libinput-debug-events`.
If you could get a row of beginning with `GESTURE_SWIPE_BEGIN`, please paste the output of `libinput-debug-events`.(if you find it, you may be able to use fusuma.)

closed 2018-04-22 by stale[bot]

juboba, 2019-06-02:
Hey guys, I'm having the same problem in a Dell XPS.

```
I, [2019-06-03T00:37:59.776545 #18053]  INFO -- : ---------------------------------------------
I, [2019-06-03T00:37:59.776572 #18053]  INFO -- : Fusuma: 0.11.1
I, [2019-06-03T00:37:59.777947 #18053]  INFO -- : libinput: 1.10.4
I, [2019-06-03T00:37:59.778505 #18053]  INFO -- : OS: Linux 4.18.0-20-generic #21~18.04.1-Ubuntu SMP Wed May 8 08:43:37 UTC 2019
I, [2019-06-03T00:37:59.778984 #18053]  INFO -- : Distribution: Ubuntu 18.04.2 LTS \n \l
I, [2019-06-03T00:37:59.779517 #18053]  INFO -- : Desktop session: ubuntu
I, [2019-06-03T00:37:59.779568 #18053]  INFO -- : ---------------------------------------------
D, [2019-06-03T00:37:59.780612 #18053] DEBUG -- : {:debug_events=>"libinput list-devices"}
D, [2019-06-03T00:37:59.785693 #18053] DEBUG -- : {:available_devices=>[]}
E, [2019-06-03T00:37:59.785724 #18053] ERROR -- : Touchpad is not found
```

```
 $ xinput
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ DELL08AF:00 06CB:76AF Touchpad            id=10   [slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad                id=15   [slave  pointer  (2)]
⎜   ↳ MX Vertical Mouse                         id=16   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Video Bus                                 id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Sleep Button                              id=8    [slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate           id=9    [slave  keyboard (3)]
    ↳ Intel HID events                          id=11   [slave  keyboard (3)]
    ↳ Intel HID 5 button array                  id=12   [slave  keyboard (3)]
    ↳ Dell WMI hotkeys                          id=13   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=14   [slave  keyboard (3)]

```

My output of `libinput-debug-events` while swiping with 3 fingers up:

```
-event16  GESTURE_SWIPE_BEGIN  +2.01s   3
 event16  GESTURE_SWIPE_UPDATE  +2.01s  3 -0.72/-6.09 (-2.19/-18.59 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.02s  3 -0.81/-6.49 (-2.19/-17.50 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.03s  3  0.00/-8.11 ( 0.00/-21.87 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.04s  3 -0.41/-9.33 (-1.09/-25.15 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.04s  3 -0.41/-10.14 (-1.09/-27.34 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.05s  3  0.00/-10.14 ( 0.00/-27.34 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.06s  3  0.00/-9.74 ( 0.00/-26.25 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.06s  3 -0.41/-9.74 (-1.09/-26.25 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.07s  3  0.00/-9.74 ( 0.00/-26.25 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.08s  3  0.00/-10.55 ( 0.00/-28.43 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.08s  3 -0.81/-9.74 (-2.19/-26.25 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.09s  3 -0.41/-8.93 (-1.09/-24.06 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.10s  3 -0.81/-8.52 (-2.19/-22.97 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.10s  3 -0.41/-9.33 (-1.09/-25.15 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.11s  3 -0.81/-8.52 (-2.19/-22.97 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.12s  3 -0.41/-4.87 (-1.09/-13.12 unaccelerated)
 event16  GESTURE_SWIPE_UPDATE  +2.13s  3 -0.41/-3.25 (-1.09/-8.75 unaccelerated)
 event16  GESTURE_SWIPE_END  +2.15s     3
```

iberianpig, 2019-06-10:
Thank you!
It seems that gesture is available on your device.

BTW, have you already added $USER to input group then restart session?

If it doesn't fix the problem, please paste here the output of `libinput-list-devices | grep -A 20 -B 2 "event16"`.

I will check why fusuma cannot detect it.

juboba, 2019-06-10:
aww dammit! I thought I cancelled this. Yes, I just needed to add my user to the input group.
Thanks anyway @iberianpig Fusuma is working perfectly!