iberianpig/fusuma #57: E, [2018-01-09T14:24:33.246602 #7474] ERROR -- : Touchpad is not found

theZulqarnain, 2018-01-09:
when i run `fusuma` command it shows me error

**E, [2018-01-09T14:24:33.246602 #7474] ERROR -- : Touchpad is not found**

how to solve...?

iberianpig, 2018-01-09:
Please paste the output of 3 fingers swipes with running `libinput-debug-events`.
If you could get a row of beginning with `GESTURE_SWIPE_BEGIN`, please paste the output of `libinput-list-devices`.(if you find it, you may be able to use fusuma.)

theZulqarnain, 2018-01-10:
can explain me again,i didn't get.............?

Fabi0Z, 2018-01-11:
I have the same problem, nothing changes uscing `sudo fusuma`

I'm using elementary OS 0.4

thierry-fr, 2018-01-21:
Hi, 
Thanks for this wonderful tool !

I had the same error on first launch. After the creation of the configuration file in the due folder and launching fusuma with sudo, everything works perfectly.

giordanorn, 2018-01-25:
Hey, where/how did you create this configuration file? I'm having the same issue here.

thierry-fr, 2018-01-26:
Hi,

I just did :

1/
touch ~/.config/fusuma/config.yml

2/
paste the sample config in "config.yml" and save

3/
start fusuma with sudo

4/
Add fusuma to startup programs.

Thierry

zhuhaozh, 2018-03-10:
Also have this problem previously,  sudo fusuma is fine for me on Ubuntu 16.04

ievgenk, 2018-05-15:
Same here, when running fusuma without sudo gives the same error.
However when I run it with sudo fusuma, I hit enter and nothing happens.

Please help :)

areisle, 2018-06-02:
I'm having this issue as well; I was following instructions from this post https://italolelis.com/posts/multitouch-gestures-ubuntu-fusuma. I'm running ubuntu 18.04. the output from libinput-list-devices was as follows:
Device:           Power Button
Kernel:           /dev/input/event3
Group:            1
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           Video Bus
Kernel:           /dev/input/event5
Group:            2
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           Power Button
Kernel:           /dev/input/event1
Group:            3
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           Lid Switch
Kernel:           /dev/input/event0
Group:            4
Seat:             seat0, default
Capabilities:     switch
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
[... 250 lines cut ...]
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           SynPS/2 Synaptics TouchPad
Kernel:           /dev/input/event6
Group:            14
Seat:             seat0, default
Size:             70x50mm
Capabilities:     pointer gesture
Tap-to-click:     disabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *two-finger edge 
Click methods:    *button-areas clickfinger 
Disable-w-typing: enabled
Accel profiles:   none
Rotation:         n/a

iberianpig, 2018-06-17:
@areisle Could you paste outputs of `fusuma -v` after updating fusuma with `gem update fusuma`?

areisle, 2018-06-18:
`Fusuma: 0.7.2
libinput: 1.10.4
OS: Linux 4.15.0-22-generic #24-Ubuntu SMP Wed May 16 12:15:17 UTC 2018
Distribution: Ubuntu 18.04 LTS \n \l

Desktop session: ubuntu
---------------------------------------------
D, [2018-06-18T07:54:12.833873 #16597] DEBUG -- : {:debug_events=>"libinput list-devices"}
D, [2018-06-18T07:54:13.298057 #16597] DEBUG -- : {:device_names=>["event12"]}
D, [2018-06-18T07:54:13.303714 #16597] DEBUG -- : {:debug_events=>"stdbuf -oL -- libinput debug-events --device /dev/input/event12"}

`
and then this is if I then try 3 and 4 finger movements
`---------------------------------------------
Fusuma: 0.7.2
libinput: 1.10.4
OS: Linux 4.15.0-22-generic #24-Ubuntu SMP Wed May 16 12:15:17 UTC 2018
Distribution: Ubuntu 18.04 LTS \n \l

Desktop session: ubuntu
---------------------------------------------
D, [2018-06-18T07:54:12.833873 #16597] DEBUG -- : {:debug_events=>"libinput list-devices"}
D, [2018-06-18T07:54:13.298057 #16597] DEBUG -- : {:device_names=>["event12"]}
D, [2018-06-18T07:54:13.303714 #16597] DEBUG -- : {:debug_events=>"stdbuf -oL -- libinput debug-events --device /dev/input/event12"}
D, [2018-06-18T07:55:05.562915 #16597] DEBUG -- : {:time=>"+52.25s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.46"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.569940 #16597] DEBUG -- : {:time=>"+52.25s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"0.75"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.577166 #16597] DEBUG -- : {:time=>"+52.26s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"0.90"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.577446 #16597] DEBUG -- : {:x=>0.0, :y=>1.0366666666666666}
D, [2018-06-18T07:55:05.584084 #16597] DEBUG -- : {:time=>"+52.27s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.03"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.584242 #16597] DEBUG -- : {:x=>0.0, :y=>1.035}
D, [2018-06-18T07:55:05.591416 #16597] DEBUG -- : {:time=>"+52.27s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.15"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.591680 #16597] DEBUG -- : {:x=>0.0, :y=>1.0579999999999998}
D, [2018-06-18T07:55:05.598441 #16597] DEBUG -- : {:time=>"+52.28s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.25"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.598705 #16597] DEBUG -- : {:x=>0.0, :y=>1.0899999999999999}
D, [2018-06-18T07:55:05.605814 #16597] DEBUG -- : {:time=>"+52.29s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.00"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.606080 #16597] DEBUG -- : {:x=>0.0, :y=>1.077142857142857}
D, [2018-06-18T07:55:05.612662 #16597] DEBUG -- : {:time=>"+52.30s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.03"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.613032 #16597] DEBUG -- : {:x=>0.0, :y=>1.0712499999999998}
D, [2018-06-18T07:55:05.619564 #16597] DEBUG -- : {:time=>"+52.30s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.07"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.619822 #16597] DEBUG -- : {:x=>0.0, :y=>1.071111111111111}
D, [2018-06-18T07:55:05.626717 #16597] DEBUG -- : {:time=>"+52.31s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.10"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.627028 #16597] DEBUG -- : {:x=>0.0, :y=>1.0739999999999998}
D, [2018-06-18T07:55:05.633856 #16597] DEBUG -- : {:time=>"+52.32s", :action=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.00", :y=>"1.13"}, :zoom=>nil}}
D, [2018-06-18T07:55:05.634166 #16597] DEBUG -- : {:x=>0.0, :y=>1.0790909090909089}
D, [2018-06-18T07:55:05.660086 #16597] DEBUG -- : {:time=>"+52.34s", :action=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2018-06-18T07:55:20.565322 #16597] DEBUG -- : {:time=>"+67.25s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"4", :directions=>{:move=>{:x=>"1.64", :y=>"0.00"}, :zoom=>"1.00"}}
D, [2018-06-18T07:55:20.572263 #16597] DEBUG -- : {:time=>"+67.26s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"4", :directions=>{:move=>{:x=>"1.13", :y=>"0.00"}, :zoom=>"1.00"}}
D, [2018-06-18T07:55:20.579342 #16597] DEBUG -- : {:time=>"+67.26s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"4", :directions=>{:move=>{:x=>"1.04", :y=>"-0.35"}, :zoom=>"1.00"}}
D, [2018-06-18T07:55:20.579572 #16597] DEBUG -- : {:diameter=>0.0}
D, [2018-06-18T07:55:20.586779 #16597] DEBUG -- : {:time=>"+67.27s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"4", :directions=>{:move=>{:x=>"1.18", :y=>"0.00"}, :zoom=>"1.00"}}
D, [2018-06-18T07:55:20.587412 #16597] DEBUG -- : {:diameter=>0.0}
D, [2018-06-18T07:55:20.593780 #16597] DEBUG -- : {:time=>"+67.28s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"4", :directions=>{:move=>{:x=>"1.78", :y=>"-0.44"}, :zoom=>"1.00"}}
D, [2018-06-18T07:55:20.594053 #16597] DEBUG -- : {:diameter=>0.0}
D, [2018-06-18T07:55:20.600994 #16597] DEBUG -- : {:time=>"+67.28s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"4", :directions=>{:move=>{:x=>"2.00", :y=>"-0.50"}, :zoom=>"1.00"}}
D, [2018-06-18T07:55:20.601270 #16597] DEBUG -- : {:diameter=>0.0}
[... cut ...]
D, [2018-06-18T07:55:20.841635 #16597] DEBUG -- : {:time=>"+67.53s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"4", :directions=>{:move=>{:x=>"0.57", :y=>"0.00"}, :zoom=>"0.99"}}
D, [2018-06-18T07:55:20.841825 #16597] DEBUG -- : {:diameter=>-0.010625000000000107}
D, [2018-06-18T07:55:20.952204 #16597] DEBUG -- : {:time=>"+67.64s", :action=>"GESTURE_PINCH_END", :finger=>"4", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}

`
I'm assuming this means it's working now but I won't have time to test it out further until tmrw

theeleventhname, 2018-06-19:
it's my setting from the website.it works,and everybody can have a view of it:
"swipe:
    3:
        up:
            shortcut: 'ctrl+super+Up'
        down:
            shortcut: 'ctrl+super+Down'
        right:
            shortcut: 'super+w'
        left:
            shortcut: 'super+shift+w'
    4:
        up:
            shortcut: 'ctrl+alt+Up'
        down:
            shortcut: 'ctrl+alt+Down'
        left:
            shortcut: 'ctrl+alt+Left'
        right:
            shortcut: 'ctrl+alt+Right'
pinch:
    in:
        shortcut: 'ctrl+plus'
    out:
        shortcut: 'ctrl+minus'
threshold:
    swipe: 1
    pinch: 1
interval:
    swipe: 1
    pinch: 1"

adamsts, 2018-07-10:
I was having this same issue on Ubuntu 18.04 (touchpad is not found) until I added my user to the iput group and logged out and back in. Hope this helps someone else.

robin-edser, 2018-07-11:
+1 for **adding user to input group and then logging out and back in** before trying to run fusuma

closed 2018-09-16 by stale[bot]

deepanchal, 2020-05-28:
> try `sudo gpasswd -a $USER input`
This will add the user to the input group. Fusuma should be able to detect your touchpad after a restart.