iberianpig/fusuma #60: Failed creating new xdo instance

ReeceNee, 2018-01-18:
I, [2018-01-18T23:18:56.256810 #2029]  INFO -- : trigger keyevent: super+w
No protocol specified
Error: Can't open display: (null)
Failed creating new xdo instance
I, [2018-01-18T23:19:08.939509 #2029]  INFO -- : trigger keyevent: super+w
No protocol specified
Error: Can't open display: (null)
Failed creating new xdo instance
I, [2018-01-18T23:19:10.338026 #2029]  INFO -- : trigger keyevent: ctrl+w
No protocol specified
Error: Can't open display: (null)
Failed creating new xdo instance

ReeceNee, 2018-01-18:
In ubuntu 17.10 gnome

sourbloom, 2018-01-18:
I'm assuming you followed all of the read me? I've been using this successfully on Ubuntu 16.04 (unity) for a few days now. Maybe try running xdotool in command line to see if it's installed correctly.

ReeceNee, 2018-01-19:
Thanks for your reply @sourbloom .I did run it well in ubuntu 16.10 before, but something wrong happend in ubuntu 17.10. I have found it's a bug in xdotools , not fusuma. I will do something to check it.

ReeceNee, 2018-01-19:
i solved it by reinstall xdotools.
But it may seems that 'super' is doesn't work in xdotools.

Mr-ChangK, 2018-03-30:
hi,ReeceNee.I meet your problem and i use 'sudo apt install xdotools' reinstall xdotools, but fusuma still not work.please tell me how did it.i use ubuntu 17.10 too.thanks

ReeceNee, 2018-03-30:
In the ubuntu 17.10, if your xdotools doesn't work, you should try to switch to xorg modle.
So my final solution is switching ubuntu to xorg model when logging in the system.
reference:   https://askubuntu.com/questions/961304/how-do-you-switch-from-wayland-back-to-xorg-in-ubuntu-17-10

closed 2018-06-06 by stale[bot]