iberianpig/fusuma #63: Add rotation gesture support

MrDrMcCoy, 2018-02-05:
Would it be possible to add support for 2-finger rotation gestures? For example:

- rotate-left: 'ctrl+Tab' 
- rotate-right: 'ctrl+shift+Tab' 

This would work well for stepping through tabs in many apps.

Otherwise, Fusuma is working well for me on Solus Linux. Thank you for your wonderful project!

iberianpig, 2018-02-07:
It's a nice idea!
Could you please make a pull request?

MrDrMcCoy, 2018-02-07:
@iberianpig, I sadly lack the requisite skills in Ruby to add such functionality, and only really know Python.  :( 

@benklop, Could I lean on your skills to make this happen?

navedcoded, 2018-02-11:
To make this even simpler, I think it could be implemented as just detecting two fingers moving in opposite directions (my attempts at doing a two-finger rotation seem pretty much that).

iberianpig, 2018-02-12:
Below is a output of `libinput-debug-events`.
```
event14  GESTURE_PINCH_UPDATE +113.48s 3  0.61/-0.61 ( 1.64/-1.64 unaccelerated)  0.53 @ -0.03
event14  GESTURE_PINCH_UPDATE +113.48s 3  0.61/ 0.00 ( 1.64/ 0.00 unaccelerated)  0.53 @ -0.12
```

It gets minus values(`@ -0.03`) after `@` with counterclockwise rotation gestures.

closed 2018-04-22 by stale[bot]