iberianpig/fusuma #66: Ubuntu 18.04 ships with ruby 2.5, needs manual ruby 2.3 installation

faiv, 2018-02-25:
After upgrading to the latest 18.04 development version, and upgrading to ruby 2.5, fusuma did not work anymore. I had to manually install ruby 2.3 to run fusuma again. A symlink did not help.

# ln -s /usr/bin/ruby2.3 /usr/bin/ruby
# fusuma
Traceback (most recent call last):
	2: from /usr/local/bin/fusuma:22:in `<main>'
	1: from /usr/lib/ruby/2.5.0/rubygems.rb:263:in `bin_path'
/usr/lib/ruby/2.5.0/rubygems.rb:289:in `find_spec_for_exe': can't find gem fusuma (>= 0.a) with executable fusuma (Gem::GemNotFoundException)

olegsenko, 2018-03-01:
I have ASUS n750 laptop with ruby 2.5 installed. 
For me libinput 1.8 is working, and the latest one didn't recognize any gestures in fusuma.

iberianpig, 2018-03-05:
@faiv Why did you need to downgrade Ruby to 2.5?
Fusuma works with Ruby 2.5 on my environment.

closed 2018-03-22 by iberianpig

pockata-kt, 2018-08-21:
I'm having the same issue, ruby version `ruby 2.5.0p0 (2017-12-25 revision 61468) [x86_64-linux]`