iberianpig/fusuma #71: security issues

the0demiurge, 2018-04-13:
By adding user to "input" group, one's all input behavior are easy be monitored, which       may cause leak of password or so.

Users should be warned

iberianpig, 2018-04-13:
@the0demiurge Thanks!
Do you have any option that keep it secure from the leak?

I think below is ok
* run fusuma with `sudo fusuma` without adding user to "input" group.

whot, 2018-04-30:
You can drop a sudo config snippet to allow running fusuma without a password as sudo. IMO that's the easiest solution.

iberianpig, 2018-06-28:
@whot Do you mean that it is better to execute fusuma without sudo with configuring `sudoers` file?

whot, 2018-06-29:
you can configure sudo so you can execute specific commands without asking for passwords. You still need to run it as `sudo <command>` though

js-d-coder, 2018-07-20:
`sudo fusuma` does not works. I have config file in both `~/.config/fusuma` and `/root/.config/fusuma/`.

closed 2018-09-25 by stale[bot]