iberianpig/fusuma #76: Deprecation warning

z0rc, 2018-05-10:
### Prerequisites

* [x] Put an X between the brackets on this line if you have done all of the following:
    * Read Documents. https://github.com/iberianpig/fusuma#installation
    * Checked that `libinput-debug-events` worked correctly.
    * Reproduced the problem in latest version. Update fusuma with `gem update fusuma`
    * Checked that your issue isn't already filed: https://github.com/iberianpig/fusuma/issues

### Description

```
% libinput-debug-events       
WARNING: This compatibility wrapper will be removed in the future. Please run 'libinput debug-events' instead
```

### Steps to Reproduce

1. Run `libinput-debug-events` in console

**Expected behavior:** No deprecation warnings

**Actual behavior:** Deprecation warning

**Reproduces how often:** 100%

### Versions

fusuma: b1ab71776bc47694cd7882403913303958ff29ca
OS: Kubuntu 17.10
libinput: 1.8.4

z0rc, 2018-05-15:
It appears that `libinput debug-events` is relatively new command and isn't available in libinput present in Ubuntu 16.04. Also Ubuntu 18.04 still provides `libinput-debug-events`. Therefore this issue doesn't require immediate action and can be addressed later.

iberianpig, 2018-06-17:
fix with https://github.com/iberianpig/fusuma/commit/c399e7a710b1efed5b9a7db66582c91fa5f626e1

closed 2018-06-17 by iberianpig