iberianpig/fusuma #77: Option to filter libinput by device

z0rc, 2018-05-10:
### Prerequisites

* [x] Put an X between the brackets on this line if you have done all of the following:
    * Read Documents. https://github.com/iberianpig/fusuma#installation
    * Checked that `libinput-debug-events` worked correctly.
    * Reproduced the problem in latest version. Update fusuma with `gem update fusuma`
    * Checked that your issue isn't already filed: https://github.com/iberianpig/fusuma/issues

### Description

`libinput-debug-events` by default catches all events, in addition to touchpad events there are keyboard, mouse and more. `libinput-debug-events` allows to filter those out by using `--device` option. This filtering is allows to consume less CPU time by `fusuma`+`libinput-debug-events`. For example, when I plug mouse to laptop and move it relatively quickly in constant manner `fusuma` and `libinput-debug-events` consume about 10-15% of single CPU core. `libinput-debug-events --device ...` allows to ignore mouse events and effectively consume zero CPU time on mouse movements.

This option is a bit tricky, as user have to know device name before hand. So maybe we can add an advanced option to define `libinput-debug-events` launch parameters?

### Versions

fusuma: https://github.com/iberianpig/fusuma/commit/b1ab71776bc47694cd7882403913303958ff29ca
OS: Kubuntu 17.10
libinput: 1.8.4

iberianpig, 2018-05-31:
Thanks for your suggestion!
I will remove the device that doesn't include `Size:`.

iberianpig, 2018-06-24:
@z0rc `--device=`option issue available 🎉

closed 2018-06-25 by iberianpig