iberianpig/fusuma #78: Questionable necessity of --enable-dwt

z0rc, 2018-05-10:
### Prerequisites

* [x] Put an X between the brackets on this line if you have done all of the following:
    * Read Documents. https://github.com/iberianpig/fusuma#installation
    * Checked that `libinput-debug-events` worked correctly.
    * Reproduced the problem in latest version. Update fusuma with `gem update fusuma`
    * Checked that your issue isn't already filed: https://github.com/iberianpig/fusuma/issues

### Description

`libinput-debug-events` launched with `--enable-dwt` option and this is hardcoded. It's usually a desired option to turn it on, but at fusuma case it's meaningless. This option affects whole libinput, which should be managed by user via GNOME/KDE/whatever input configuration. Forcefully enabling it for fusuma isn't right and may lead to hard to debug issue when user decides to disable this option for various reasons.

As this option doesn't change `libinput-debug-events` output, I believe it should be dropped from fusuma.

### Versions

You can get this information from copy and pasting the output of `fusuma --version` from the command line. Also, please include the OS and what version of the OS you're running.

iberianpig, 2018-05-11:
Oh, thanks! Could you make pull request?

z0rc, 2018-05-11:
Sure, see linked PR.

PR #80 (2018-05-15): Remove --enable-dwt option
Because it's meaningless for fusuma use case and has side effect.

Closes #78.
commit 74d7c147 (2018-05-15): Remove --enable-dwt option

Because it's meaningless for fusuma use case and has side effect.

Closes #78.
closed 2018-05-15 by iberianpig