iberianpig/fusuma #83: Support for tap event

m93a, 2018-05-29:
### Prerequisites
* [X] Put an X here if you have done all the necessary stuff.

### Description
It would be quite useful if Fusuma had the ability to also detect the `tap` events. That would for example allow me to use both two-finger tap and right-corner click for the emulation of the right mouse button. AFAIK there is currently no way to achieve this behavior in Ubuntu 18.04, see my [SO question](https://askubuntu.com/questions/1041592/clickpad-how-to-right-click-with-both-two-fingers-and-area).

Adding `tap` and possibly `doubletap` with options for two, three and four fingers would IMHO greatly extend the capabilities of Fusuma.

iberianpig, 2018-05-30:
@m93a 
fusuma depends on libinput-debug-events.
Unfortunately, libinput-debug-events doesn't recognize tapping and coordinate data(x, y) on a touchpad.

closed 2018-05-30 by iberianpig

iberianpig, 2020-03-20:
@m93a  I released fusuma-plugin-tap.
This plugin detects 1~4 fingers tap gestures and hold(long press) gestures.