iberianpig/fusuma #86: libinput-debug-events blocking USB hotplug

MisterPresident, 2018-06-19:
Hi,
I am having a problem with my mouse/keyboard hotplug. Thanks to the people [@archlinuxforum](https://bbs.archlinux.org/viewtopic.php?pid=1792649) i found out that
 "libinput-debug-events" started by fusuma doesn't allow hot-plugging my input devices.

**Scenarios**
_Scenario 1_
Start Laptop no mouse plugged in - ligthdm greeter - plug the mouse in (working) - login (starting fusuma) - mouse still working - plug out - plug in (in dmesg the new hardware shows up) -  but it's not working

_Scenario 2_
Start Laptop no mouse plugged in - ligthdm greeter - login (starting fusuma) - plug the mouse in (working) - plug out - plug in - still working

**Reproduce**
evtest (devices working) - plug out (e.g. keyboard) - plug in - evtest (devices not working because they are blocked with "libinput-debug-events") - kill this process - evtest (devices working) - plug out - plug in - evtest (devices still working)  - .. - working forever

**Versions:**
Fusuma: 0.7.2
libinput: 1.11.0
OS: Linux 4.16.13-2-ARCH #1 SMP PREEMPT Fri Jun 1 18:46:11 UTC 2018
Distribution: Arch Linux
Desktop session: i3

iberianpig, 2018-06-28:
@MisterPresident H
This is libinput-debug-events issue.
What do you think how do it fix this problem on fusuma?

MisterPresident, 2018-07-02:
You are right. libinput 1.11.1 fixed this issue.

closed 2018-07-02 by MisterPresident