iberianpig/fusuma #87: libinput 1.11.1: ERROR -- : Touchpad is not found

tobac, 2018-06-19:
### Description

Arch Linux updated their libinput to version 0.11.1. After said update, fusuma refuses to start and gives the following error:

`ERROR -- : Touchpad is not found`

### Versions
```
---------------------------------------------
Fusuma: 0.7.2
libinput: 1.11.1
OS: Linux 4.17.2-1-ARCH #1 SMP PREEMPT Sat Jun 16 11:08:59 UTC 2018
Distribution: Arch Linux \r (\l)

Desktop session: /usr/share/xsessions/plasma
---------------------------------------------
E, [2018-06-19T20:15:16.991999 #10655] ERROR -- : Touchpad is not found
```

iberianpig, 2018-06-19:
please paste the output of libinput-list-devices.

tobac, 2018-06-19:
"libinput list-devices" segfaults. I just saw a thread at the Arch Linux forums describing the very same problems, so it might be an Arch issue rather than a Fusuma issue: https://bbs.archlinux.org/viewtopic.php?pid=1792728

tobac, 2018-06-20:
Update: Someone opened a GitLab issue for libinput: https://gitlab.freedesktop.org/libinput/libinput/issues/50 which led to https://gitlab.freedesktop.org/libinput/libinput/commit/7c51c881dc72827ca50ae9eea3d123eef9cb4a66

Therefore things should be up and running again once the next minor version of libinput is released.

closed 2018-08-26 by stale[bot]

iberianpig, 2018-12-04:
It's cannot detect available touchpad. 

Please paste output of `fusuma -l`

BTW, it does not have to do with "sudo".
Fusuma needs access privileges on the input group, "sudo" just bypass it with root access.

niliushall, 2018-12-04:
Maybe I just didn`t give it access to using the touchpad.

There is the output of two methods:

### 1. fusuma -l  
```
E, [2018-12-04T21:36:10.664085 #9272] ERROR -- : Touchpad is not found
```

### 2. sudo fusuma -l
```
USB Keyboard
USB Optical Mouse
ETPS/2 Elantech Touchpad
```