iberianpig/fusuma #88: Support multi finger taps

akabhirav, 2018-06-29:
# Support for multifinger taps

 Multifinger taps are very useful, one example that I can see on Windows is 4 tap to open the action center(basically  a glorified sidebar with all the notifications and quick settings) 

Supporting multifinger taps such as 3, 4, 5  finger taps which are not already preconfigured would be helpful in a similar way. Some use cases are opening up notifications by triggering `xdotool super+m` or something similar

deb75, 2018-07-02:
Hello,

My question is very close to this one :  does fusama works with touchscreen ? Or is it implemented with only touchpads in mind ?

Multifinger taps are what one may want to use on a touchscreen.

Regards

iberianpig, 2018-07-02:
@akabhirav
Libinput-debug-events doesn't detect a touch. As a result, touch is not supported and will only support swiping, pinching, and rotating.

@deb75 
I don't have touchscreen device.
So, please confirm supportability on touchscreen with libinput-debug-events.
Then please write code and make pull request to support touchscreen.

deb75, 2018-07-02:
Hi

Thanks for your answeer.

My laptop is a surface book 2 equipped with a touchscreen.

When running `libinput debug-events` and touching the screen, I see `TOUCH_DOWN`, when I leave my finger, it says `TOUCH_UP`.

Both previous events are displayed for each finger touching the screen and of course it detects the fingers motions.

So I guess that by writing a code similar to `libinput debug-events` I could retrieve these events and the finger coordinates so as to determine whether it is a 2,3,4, ... fingers touch, and if the motion is swiping, pinching, ...

Is that what you meant by writing a code ?

Regards

iberianpig, 2018-07-03:
@deb75 Thanks!
It become clear libinput-debug-events can detects multi-touch on touchscreen with your reports.
But I don't have touch screen devices.
So, I wish that you or other person that have touchscreen devices make pull-request for touchscreen support.

akabhirav, 2018-07-23:
@iberianpig Hey, I didn't mean taps on a touch-screen, I meant finger down on a touchpad. I just wanted features with something like multiple taps, for example: 4 finger down on touchpad would open notifications

iberianpig, 2018-07-23:
@akabhirav  
 OK, I have understood what you mean.
`libinput-debug-events` don't detect tap events on Touchpad.
So, Fusuma also can't detect tap events. :cry:

closed 2018-09-28 by stale[bot]

dhc1, 2018-12-05:
libinput-debug-events --enable-tap can detect 1, 2,3 finger taps.
We can tap feature with it .

iberianpig, 2020-03-20:
@dhc1 @akabhirav @deb75 

I released fusuma-plugin-tap.
This plugin detects 1~4 fingers tap gestures and hold(long press) gestures.

akabhirav, 2020-03-20:
When I started using fusuma I had just moved back from macOS to Linux. I really loved fusuma and believed(and still do) that this is the best solution out there that can match the touchpad gestures in macOS. Great work and thanks for implementing the feature.