iberianpig/fusuma #89: Show all windows takes a while to execute

vinyoliver, 2018-07-19:
Hello, 
All gestures wosks just fine but whenever I try to use three fingers up/down to show all windows, always takes around 2/3 seconds to reproduce the command. After the first time if I do it again works fine. But if I try it again later the same thing happens. If I just press the super class works like charm

I'm on ubuntu 18.04 with default gnome.

Fwardalonious, 2018-08-09:
I'm having the same issue as you vinyoliver, I'm a bit of a Ubuntu newb so I've been looking for help on this too.

vinyoliver, 2018-08-09:
Some people reported the same problem here: https://italolelis.com/posts/multitouch-gestures-ubuntu-fusuma/ I believe it's something happening to multiple users. Hope to get a solution soon in the future.

Fwardalonious, 2018-08-10:
Actually I found this page from that article but for whatever reason disqus kept marking my comments as spam so I don't think you ever saw my original post. I do too, I recently switched to Linux on my laptop and I've been slowly working through each issue and I'm still stuck on this one.

iberianpig, 2018-08-26:
Which way do you use wayland and X11?
Can you consider using other keys than the super key?
 Even with other tools like xbindkeys, it seems that detection of a super key press is also slow.

Fwardalonious, 2018-08-26:
I use x11 and with the super key it's buttery smooth, why what are you thinking?

iberianpig, 2018-08-26:
I doubt that window manager will prevent sending the first custom shortcut including a super key. 

Because xbindkeys also didn't not recognized a custom shortcut including a super key at first. It require twice.

I checked on Pantheon (X11) and Gnome shell(X11).

vinyoliver, 2018-09-02:
I've been using fusuma for a while now and I realized that it's not just the show all windows command that takes a while to execute. Pretty much every gesture I try takes a while to execute for the first time on Ubuntu 18.04. If I use one gesture after another works great but if I don't use for a while and try again the same thing happens.

iberianpig, 2018-09-02:
I think this problem caused by xdotool and Ubuntu 18.04, so we may need to find other command as workaround instead of xdotool. 😥

Fwardalonious, 2018-09-03:
It seems to be, I can't seem to figure out what triggers the sluggishness because about 20%-30% of the time it's smooth....but unfortunately not fast enough enough of the time to let me use it as intended. I wish I knew more to be able to help fix it. Any more info I can send you to help you figure out the issue?

Fwardalonious, 2018-09-03:
Ok....for example right now it's working wonderfully. I don't know why, but in a bit it'll get slow again. Very odd.

closed 2018-11-09 by stale[bot]