iberianpig/fusuma #91: Does 'AlpsPS/2 ALPS DualPoint TouchPad' support Muli-Touch？

SunYongshuai, 2018-07-27:
**Describe the bug**
Touchpad Information:
Device:           AlpsPS/2 ALPS DualPoint TouchPad
Kernel:           /dev/input/event8
Group:            7
Seat:             seat0, default
Size:             97.50x53.87mm
Capabilities:     pointer 
Tap-to-click:     disabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *two-finger edge 
Click methods:    *button-areas clickfinger 
Disable-w-typing: enabled
Accel profiles:   none
Rotation:         n/a

And I use 'fusuma -v', I get:
D, [2018-07-27T18:08:33.532891 #12585] DEBUG -- : {:diameter=>-0.08071428571428568}
D, [2018-07-27T18:08:33.544756 #12585] DEBUG -- : {:time=>"+9.03s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"2", :directions=>{:move=>{:x=>"-0.07", :y=>"-0.16"}, :zoom=>"0.32"}}
D, [2018-07-27T18:08:33.545037 #12585] DEBUG -- : {:diameter=>-0.08172413793103445}
D, [2018-07-27T18:08:33.556593 #12585] DEBUG -- : {:time=>"+9.05s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"2", :directions=>{:move=>{:x=>"0.00", :y=>"2.25"}, :zoom=>"0.32"}}
D, [2018-07-27T18:08:33.556883 #12585] DEBUG -- : {:diameter=>-0.08266666666666661}
D, [2018-07-27T18:08:33.569244 #12585] DEBUG -- : {:time=>"+9.06s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"2", :directions=>{:move=>{:x=>"0.00", :y=>"-0.35"}, :zoom=>"0.32"}}
D, [2018-07-27T18:08:33.569568 #12585] DEBUG -- : {:diameter=>-0.08354838709677415}
D, [2018-07-27T18:08:33.580802 #12585] DEBUG -- : {:time=>"+9.07s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"2", :directions=>{:move=>{:x=>"0.00", :y=>"-0.66"}, :zoom=>"0.32"}}
D, [2018-07-27T18:08:33.581156 #12585] DEBUG -- : {:diameter=>-0.08437499999999992}
D, [2018-07-27T18:08:33.592225 #12585] DEBUG -- : {:time=>"+9.08s", :action=>"GESTURE_PINCH_UPDATE", :finger=>"2", :directions=>{:move=>{:x=>"0.00", :y=>"-0.83"}, :zoom=>"0.32"}}
D, [2018-07-27T18:08:33.592522 #12585] DEBUG -- : {:diameter=>-0.08515151515151509}
D, [2018-07-27T18:08:33.658893 #12585] DEBUG -- : {:time=>"+9.15s", :action=>"GESTURE_PINCH_END", :finger=>"2", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}

Only finger=>"2" shown in the output.(Zoom in and zoom out functions)
When I test three fingers, Nothing output.

Thanks a lot!

iberianpig, 2018-07-30:
Did you already check that `libinput-debug-events`  didn't print swipe event?

SunYongshuai, 2018-07-30:
It just gives some movement's details as follow: 
event8   POINTER_AXIS      +7.28s	vert 6.71* horiz 14.06* (finger)
 event8   POINTER_AXIS      +7.29s	vert 0.96* horiz 3.12* (finger)
 event8   POINTER_AXIS      +7.31s	vert 7.67* horiz 4.86* (finger)
 event8   POINTER_AXIS      +7.32s	vert 2.30* horiz 17.53* (finger)
 event8   POINTER_AXIS      +7.33s	vert 9.21* horiz 6.94* (finger)
 event8   POINTER_AXIS      +7.34s	vert 2.69* horiz 5.55* (finger)
 event8   POINTER_AXIS      +7.35s	vert 9.21* horiz 15.10* (finger)
 event8   POINTER_AXIS      +7.37s	vert 3.07* horiz 3.12* (finger)
 event8   POINTER_AXIS      +7.38s	vert 2.49* horiz 13.54* (finger)
 event8   POINTER_AXIS      +7.39s	vert 2.69* horiz 2.95* (finger)
 event8   POINTER_AXIS      +7.40s	vert 8.06* horiz 3.64* (finger)
 event8   POINTER_AXIS      +7.41s	vert 1.34* horiz 15.62* (finger)
 event8   POINTER_AXIS      +7.43s	vert 1.34* horiz 5.73* (finger)
 event8   POINTER_AXIS      +7.44s	vert 1.34* horiz 5.90* (finger)
 event8   POINTER_AXIS      +7.45s	vert 6.91* horiz 16.49* (finger)
 event8   POINTER_AXIS      +7.46s	vert 0.77* horiz 4.51* (finger)
 event8   POINTER_AXIS      +7.47s	vert 6.71* horiz 15.10* (finger)
 event8   POINTER_AXIS      +7.49s	vert 0.58* horiz 3.30* (finger)
 event8   POINTER_AXIS      +7.50s	vert 1.15* horiz 3.12* (finger)
 event8   POINTER_AXIS      +7.51s	vert 7.48* horiz 13.54* (finger)
 event8   POINTER_AXIS      +7.52s	vert 0.96* horiz 2.26* (finger)
 event8   POINTER_AXIS      +7.53s	vert 0.58* horiz 1.39* (finger)
 event8   POINTER_AXIS      +7.54s	vert 0.19* horiz 0.52* (finger)
 event8   POINTER_AXIS      +7.58s	vert 0.00 horiz -0.35* (finger)
 event8   POINTER_AXIS      +7.59s	vert -0.19* horiz -0.17* (finger)
 event8   POINTER_AXIS      +7.62s	vert 0.00 horiz -0.17* (finger)
 event8   POINTER_AXIS      +7.63s	vert -0.19* horiz 0.00 (finger)

and I don't know whether sweep it is. And I also don't know how many figures it shows.

And some times it give an error as follow: 

[image]

SunYongshuai, 2018-07-30:
My Platform is
Ubuntu18.04.1(Linux xxx 4.15.0-29-generic #31-Ubuntu SMP Tue Jul 17 15:39:52 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux)

iberianpig, 2018-08-17:
It will get output like below if your environment is available to detect swipe.

```
 event10  GESTURE_SWIPE_BEGIN  +1.52s   3
 event10  GESTURE_SWIPE_UPDATE  +1.52s  3 15.05/-0.36 (45.93/-1.09 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.53s  3 19.07/-0.81 (51.40/-2.19 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.54s  3 21.10/-0.81 (56.87/-2.19 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.54s  3 20.69/-0.81 (55.77/-2.19 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.55s  3 20.29/-1.22 (54.68/-3.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.56s  3 19.47/-1.22 (52.49/-3.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.57s  3 17.04/-0.81 (45.93/-2.19 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.57s  3 16.23/-0.81 (43.74/-2.19 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.58s  3 13.79/-0.81 (37.18/-2.19 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.59s  3 12.17/-1.22 (32.81/-3.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.59s  3 10.55/-1.22 (28.43/-3.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.60s  3  9.74/-1.22 (26.25/-3.28 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.61s  3  8.11/-1.62 (21.87/-4.37 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.62s  3  6.09/-1.62 (16.40/-4.37 unaccelerated)
 event10  GESTURE_SWIPE_UPDATE  +1.62s  3  4.87/-0.81 (13.12/-2.19 unaccelerated)
 event10  GESTURE_SWIPE_END  +1.64s     3
```

closed 2018-10-23 by stale[bot]