iberianpig/fusuma #94: Alt+Escape and alt+shift+Escape don't do what I'd expect

ptoner, 2018-08-22:
I'm using the following configuration (partial):

`
  3: 
    left: 
      command: 'xdotool key alt+Escape'
    right: 
      command: 'xdotool key alt+shift+Escape'
`

These do mostly what I'd expect but there's a small but significant difference between these actions and what happens when you manually press these key configurations. It re-displays the windows in seemingly random order. 

If I look at the fusumu event log I see the correct output. But it definitely doesn't have the same behavior as manually pressing the buttons. 

Can anyone think of why that might happen?

ptoner, 2018-08-23:
Update:

It seems like capitalizing Alt makes it work the way I'd expect. Not sure why. Also not positive this is exactly true yet but using it for a minute or two it seems better.

```
  3: 
    left: 
      command: 'xdotool key Alt+Escape'
    right: 
      command: 'xdotool key Alt+Escape+shift'

```

ptoner, 2018-08-23:
Another update:

Didn't really work. Only switches back and forth between 2 apps. Not sure why that would be either.

iberianpig, 2018-08-23:
Could you test with only xdotool?
I want to know this issue caused by fusuma.

ptoner, 2018-08-23:
Ok, I did. I opened a bunch of windows and then did a bunch of 3 finger swipes in a row and it would only cycle between the last 2 windows.

Then I ran the following script and it cycled through all of them over and over until I had to reboot. Which is what I'd expect since I forgot to actually increment COUNTER. But the xdotool command worked as expected.

I'm on Ubuntu 18.04.

```
#!/bin/bash 
COUNTER=0
while [  $COUNTER -lt 5 ]; do
    xdotool key alt+Escape
done
```

ptoner, 2018-09-11:
I'm not sure exactly what you mean, but I get the same behavior with when I use both the default ubuntu login or switch to using Wayland.

iberianpig, 2018-09-12:
Thanks.

Please try following config

```
command: xdotool keydown alt && sleep .1 && xdotool key Tab  && xdotool keyup alt
```
 
It works for me.

BTW, I found that switching window with "`xdotool key alt+tab`" need focus before executing.

ptoner, 2018-09-12:
Oh, gotcha. I think the reason it's not working the way I expected is because when I'm manually hitting alt-enter over and over I'm actually holding down the alt key the entire time. Seeing your code made me realize that's why the behavior is different. 

What I'm looking for is a way for swiping left to predictably cycle through the open windows and swiping right to predictably cycle the other way through the list. 

Do you think that's possible? Now that I see what's happening maybe it's not.

iberianpig, 2018-09-16:
I think it is difficult.

Because it needs to execute three different steps in order.
1. `keydown alt`
2. `send Tab` (repeat)
3. `keyup alt`

Fusuma doesn't recognize 2 or more than continuous gestures.

ptoner, 2018-09-16:
That makes sense. I appreciate your help.

closed 2018-11-22 by stale[bot]