iberianpig/fusuma #96: Unable to run Fusuma on Arch: command not found

navtojchahal, 2018-08-30:
I installed fusuma however am not able to run it. I couldn't find any solutions through "googling". What should I do?

My Setup: Acer Laptop - Arch - Gnome - Xorg

[image]

iberianpig, 2018-08-31:
Could you try `fusuma -v` and paste output here?

navtojchahal, 2018-08-31:
[image]

I got the "fusuma" command working in terminal once today by messing around with the $PATH commands. But it went back to "command not found" when I reopened the terminal.

navtojchahal, 2018-08-31:
This is how I got it working temporarily..
[image]

iberianpig, 2018-08-31:
It seems that Gem installation path can not be read.
Please confirm your Ruby environment.

navtojchahal, 2018-08-31:
[image]

iberianpig, 2018-08-31:
You may read (https://wiki.archlinux.org/index.php/Ruby) and try set up again.
Then reload your ~/.bashrc.

navtojchahal, 2018-09-10:
Okay thanks, I got it working.

I added >> PATH="$PATH:$(ruby -e 'print Gem.user_dir')/bin" << to "/etc/bash.bashrc" using nano in terminal and rebooted. It's been working great since then.

closed 2018-09-10 by iberianpig