iberianpig/fusuma #97: Fusuma stops working after suspend when four finger gestures are enabled

IjonFryderyk, 2018-08-31:
Hi, Fusuma is great :)! I've been waiting for multi touch in Ubuntu for a long time. 
I have some issues with four finger gestures though. Sometimes four finger gestures are recognized as pinches. But most bothering is that after using four finger gestures Fusuma doesn't work after resuming from suspend. When four finger gestures and pinches are disabled Fusuma works without problem after resuming from sleep. Sometimes there are also some input artifacts when I use three finger gestures. For example when I use Libre Office and I switch to another work space (three finger swipe down) sentences in Libre switch places. I also use three fingers swipe right to ctrl+super+d (show desktop). When I use this gesture with Atom opened, it executes not only the command, but also inputs letter "d" in Atom. I'm on thinkpad x240 and Ubuntu 18.04. 
Is there any command which allows restart Fusuma or touchpad drivers?

iberianpig, 2018-09-01:
Thank you for creating an issue about fusuma.

Please try increasing pinch in and pinch out thresholds.
Increasing the threshold may reduce false positives.

Even if you replace Command assigned 4 fingers with other command, does not it work after reboot?
Please check the above to know whether the cause is xdotool or fusuma.

IjonFryderyk, 2018-09-01:
Thanks for your quick replay. I will try to test your solutions in a few days. I have a lot of work now and stable Fusuma is very useful, I don't want to brake it again :).

IjonFryderyk, 2018-09-07:
Unfortunately, I had to limit Fusuma configuration to two gestures – three fingers swipe up and down – super key. With all other shortcuts that contain ctrl or alt Fusuma is unstable. All those gestures sometimes execute all sorts of unwanted behaviors (from zooming like with ctrl + two finger scroll, to imputing parts of shortcut elements) or just don't work at first attempt. These flaws sometimes reappear constantly in one session but after rebooting in another session everything works much better and malfunctions are rare. I also confirm the alt+Esc command issue. 
Replacing command assigned four fingers didn't help.
Unfortunately, despite my attempts to achieve most stable configuration, Fusuma still sometimes doesn't work after suspend. Is there any way to restart Fusuma?

iberianpig, 2018-09-08:
Please kill the fusuma process by ´ps´ and ´kill´,
then start fusuma by ´fusuma --daemon --config=path/to/your/config.yml´

I'm sorry, I did not know the cause is in xdotool. 
Please paste your config.yml and autostart settings

IjonFryderyk, 2018-09-08:
My config file looks like this right now (I had disabled all commands that was causing problems): 

swipe:
  3:
#    left: 
#      command: 'xdotool key ctrl+alt+0'
#    right: 
#      command: 'xdotool key alt+escape'
    up:
      command: 'xdotool key super'
    down:
      command: 'xdotool key super'
#  4:
#    left: 
#      command: 'xdotool key ctrl+super+D'
#    right: 
#      command: 'xdotool key ctrl+super+D'
#    up: 
#      command: 'xdotool key ctrl+alt+Up'
#    down: 
#      command: 'xdotool key ctrl+alt+Down'
#pinch:
#  in:
#    command: 'xdotool key ctrl+plus'
#  out:
#    command: 'xdotool key ctrl+minus'

threshold:
  swipe: 0.4
  pinch: 1

interval:
  swipe: 0.8
  pinch: 0.1

My autostart settings is just command: fusuma in Startup Application. Should it be direct path to Fusuma?

iberianpig, 2018-09-10:
Could you try `-d` (or `--daemon`) option in Startup Application Setting?

If it doesn't work, please try following for debugging,

 1. remove fusuma from Startup Application
 2. run `fusuma -v` in terminal (print detail information for debugging)
 3. suspend
 4. try some swipes and check that fusuma is not working
 5. check whether the output on terminal has errors

iberianpig, 2018-09-10:
Then, I found the article about Thinkpad issue for suspend.

https://bugzilla.redhat.com/show_bug.cgi?id=1560723#c8
https://bbs.archlinux.org/viewtopic.php?id=229903

iberianpig, 2018-09-16:
> Fusuma works properly after suspend :). 

Good! 

> Unfortunately four finger gestures are still executing unwanted behaviors – zoom out and zoom in.

Please try increase pinch `threshold` and try the following config.

```
pinch:
  2:
    in:
      command: 'xdotool key ctrl+plus'
    out:
      command: 'xdotool key ctrl+minus'
 ```

This config is only set to 2 finger pinches but not to set 4 finger pinches

IjonFryderyk, 2018-09-19:
Hi, I did change pinch threshold. I also had disabled pinch to zoom, so I assume that this behavior isn't connected with Fusuma pinch to zoom or pinch threshold. This behavior takes place when I swipe up to change workspace to upper one but I am already on the first workspace. So only the workspace preview appears and if I'm in Libre Office or Atom the text size increases. 
Unfortunately Fusuma stopped working after suspend three times so far. The process is running but Fusuma isn't working. I failed to start Fusuma from command line. I had to restart system each time.

IjonFryderyk, 2018-10-15:
Hi, 
I found out that a way to put gestures back to life is to restart touchpad drivers:
sudo modprobe -r psmouse
sudo modprobe psmouse

So I think that it makes it clear that it is drivers/hardware problem not with Fusuma. 

Thanks for your help.

iberianpig, 2018-10-16:
I am glad you found the solution, congratulations!

closed 2018-10-17 by iberianpig

elvisisvan, 2023-03-27:
> Please kill the fusuma process by ´ps´ and ´kill´, then start fusuma by ´fusuma --daemon --config=path/to/your/config.yml´
> 
> I'm sorry, I did not know the cause is in xdotool. Please paste your config.yml and autostart settings

is fusuma daemon named "fusuma"? i can't find it in process tree, i start with "fusuma -d" and now there's no fusuma command to stop it, i think there should be "fusuma -s" as a stop flag