iberianpig/fusuma #99: Swipe va multi finger scrolling

mahajanudit, 2018-09-10:
Hi, I wanted to emulate windows 10 gestures and Mac gestures using fusuma but I am unable to do so. Here is what I have found, correct me if I am wrong.
1. Fusuma identifies swipe gestures but not continuous scrolling. For example, 3 finger scroll up/down to control volume won't work. Or 3 finger scroll left/right won't work. It only recognised swipe which is triggered once and not continuously. 
2. I wanted to emulate 3 finger drag, like in Mac. Where if you drag using 3 finger, it essentially emulates click and drag.

mahajanudit, 2018-09-10:
Title is "Swipe vs multi finger scrolling"

iberianpig, 2018-09-16:
> Fusuma identifies swipe gestures but not continuous scrolling.

Yes, this is right.
Fusuma checks if enough gesture event to execute a command in per unit time.
Different from continuous scroll events, the command is executed once in per unit time.

> emulate 3 finger drag, like in Mac. 

So the previous problem, It is difficult to recognize two or more than continuous gestures.

F-inzaghi, 2018-11-13:
Hi, is that possible develop a better version fusuma so that it can recognize scroll gestures? Like, in win10, use 3 figure to scroll left/right it can switch between applications, and if you hold on 3 finger on touchpad, it will keep showing all applications until you choose the app you want and separate your finger with touchpad.
I tried the following config, but it's not exactly the result I want.
```
swipe:
  3:
    left:
      command: 'xdotool key alt+Tab'
    right:
      command: 'xdotool key alt+Tab'
```

iberianpig, 2018-11-14:
Alt + tab has three operation processes.

1. Keep pressing alt.
2. Press a tab.
3. Stop keeping alt key pressed. 

This may be difficult with only xdotool. 

Make wrapper script that xdotool execution state. 

Make fusuma assign commands to start/continue/end status.

closed 2019-01-22 by stale[bot]