iberianpig/fusuma #104: Issue in swipe with 2 fingers

sukheja-varun, 2018-10-31:
I am trying to use gesture with 2 fingers to go back and next in browser
Below is my config file;

`swipe:
  2:
    left: 
      command: 'xdotool key alt+Left'
    right: 
      command: 'xdotool key alt+Right' 
  3:
    left: 
      command: 'xdotool key shift+Left'
    right: 
      command: 'xdotool key shift+Right' 
    up: 
      command: 'xdotool key super+w'
    down: 
      command: 'xdotool key super+a'
  4:
    left: 
      command: 'xdotool key ctrl+alt+Right'
    right: 
      command: 'xdotool key ctrl+alt+Left'
    up: 
      command: 'xdotool key ctrl+alt+Down'
    down: 
      command: 'xdotool key ctrl+alt+Up'
pinch:
  in:
    command: 'xdotool key ctrl+plus'
  out:
     command: 'xdotool key ctrl+minus'

threshold:
  swipe: 0.4
  pinch: 0.4

interval:
  swipe: 0.8
  pinch: 0.1`

iberianpig, 2018-10-31:
fusuma doesn't support 2 finger swipe because libinput driver doesn't recognize 2 fingers swipe action.

If it available, I think that two fingers swipe is difficult to use due to conflict with 2 finger scroll.

sukheja-varun, 2018-10-31:
Yes, it is difficult for 2 fingers for up and down as it conflicts with scroll but left and right can be configured as seen in Apple Mac

iberianpig, 2018-11-05:
I think that horizontal scrolling also collides with two finger swipes.
However, because libinput's driver API is not provided, I will not add features.Thank you for suggesting me.

closed 2018-11-05 by iberianpig

andrepapaz, 2019-06-18:
:cry:

hamzahamidi, 2021-12-07:
omg I'm crying here 😭