iberianpig/fusuma #112: Fusuma stops working after some time

ozym4nd145, 2019-01-15:
**Describe the bug**
Fusuma stops recognizing gestures after some time of starting it. Its more like that it goes to sleeping state and doesn't respond to any gestures. Killing the process and restarting it fixes the problem for some time, but then same thing repeats.

**To Reproduce**
Steps to reproduce the behavior:
1. Install fusuma and setup the config given below 
2. Start fusuma by `fusuma -v`
3. Continues working for 10 mins, after that it stops triggering actions on mouse events

**Expected behavior**
Fusuma continues to function without freezing after some time.

**Fusuma and OS version**
Fusuma: 0.10.2
libinput: 1.12.4
OS: Linux 4.14.91-1-MANJARO #1 SMP PREEMPT Sat Dec 29 15:40:39 UTC 2018
Distribution: Manjaro Linux \r  (\n) (\l)
Desktop session: /usr/share/xsessions/plasma

**Additional context**
config.yml:
```
swipe:
  3: 
    left: 
      command: 'xdotool key ctrl+Tab'
    right: 
      command: 'xdotool key ctrl+shift+Tab'
    up:
      command: 'xdotool key ctrl+alt+Down'
    down:
      command: 'xdotool key ctrl+alt+Up'
threshold:
  swipe: 0.5
  pinch: 1
interval:
  swipe: 0.5
  pinch: 1
```

Output from `fusuma -v`:
```
D, [2019-01-14T23:32:03.788840 #21097] DEBUG -- : {:time=>"+6968.51s", :event=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"10.19", :y=>"-3.16"}, :zoom=>nil}}
D, [2019-01-14T23:32:03.788956 #21097] DEBUG -- : {:x=>17.03833333333333, :y=>-6.3549999999999995}
D, [2019-01-14T23:32:03.789065 #21097] DEBUG -- : {:time=>"+6968.52s", :event=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"7.37", :y=>"-2.46"}, :zoom=>nil}}
D, [2019-01-14T23:32:03.789151 #21097] DEBUG -- : {:x=>15.657142857142857, :y=>-5.798571428571428}
D, [2019-01-14T23:32:03.789322 #21097] DEBUG -- : {:time=>"+6968.53s", :event=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"5.79", :y=>"-1.93"}, :zoom=>nil}}
D, [2019-01-14T23:32:03.789417 #21097] DEBUG -- : {:x=>14.42375, :y=>-5.3149999999999995}
D, [2019-01-14T23:32:03.789527 #21097] DEBUG -- : {:time=>"+6968.54s", :event=>"GESTURE_SWIPE_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"4.74", :y=>"-1.40"}, :zoom=>nil}}
D, [2019-01-14T23:32:03.789617 #21097] DEBUG -- : {:x=>13.347777777777777, :y=>-4.879999999999999}
D, [2019-01-14T23:32:03.789719 #21097] DEBUG -- : {:time=>"+6968.55s", :event=>"GESTURE_SWIPE_END", :finger=>"3", :directions=>{:move=>{:x=>nil, :y=>nil}, :zoom=>nil}}
D, [2019-01-14T23:43:51.467090 #21097] DEBUG -- : {:time=>"+7676.24s", :event=>"GESTURE_PINCH_UPDATE", :finger=>"3", :directions=>{:move=>{:x=>"0.17", :y=>"0.00"}, :zoom=>"1.00"}}
D, [2019-01-14T23:43:53.906459 #21097] DEBUG -- : {:time=>"+7678.68s", :event=>"GESTURE_PINCH_END", :finger=>"3", :directions=>{:move=>{:x=>"cancelled", :y=>nil}, :zoom=>nil}}
```
Note: fusuma process still is active but it doesn't print anything after this.

Output from libinput debug events:
```
 event12  POINTER_MOTION   +20.42s        1.05/  1.05 ( +2.00/ +2.00)
 event12  POINTER_MOTION   +20.43s        1.05/  0.53 ( +2.00/ +1.00)
 event12  POINTER_MOTION   +20.44s        0.53/  0.53 ( +1.00/ +1.00)
 event12  GESTURE_SWIPE_BEGIN +20.76s   3
 event12  GESTURE_SWIPE_UPDATE +20.76s  3 21.05/ 1.58 (78.74/ 5.91 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.77s  3 18.85/ 0.91 (68.24/ 3.28 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.78s  3 33.25/ 0.48 (90.55/ 1.31 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.79s  3 26.01/-1.22 (70.21/-3.28 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.80s  3 30.63/-3.27 (85.96/-9.19 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.81s  3 26.56/-3.83 (72.83/-10.50 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.82s  3 17.58/-3.07 (56.43/-9.84 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.83s  3 16.63/-2.83 (57.74/-9.84 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.84s  3 14.66/-3.34 (51.84/-11.81 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.85s  3  9.70/-2.29 (36.09/-8.53 unaccelerated)
 event12  GESTURE_SWIPE_UPDATE +20.86s  3  7.54/-1.93 (28.22/-7.22 unaccelerated)
 event12  GESTURE_SWIPE_END +20.87s     3
```
Note: It continues to give GESTURE_SWIPE_UPDATE after fusuma stops responding to these events.

Output of htop:
```
21098 oz         20   0 96120 13692  4972 S  0.0  0.2  0:00.00 /usr/bin/ruby /home/oz/bin/fusuma -v
21108 oz         20   0 96120 13692  4972 S  0.0  0.2  0:00.00 /usr/bin/ruby /home/oz/bin/fusuma -v
21097 oz         20   0 96120 13692  4972 S  0.0  0.2  0:09.54 /usr/bin/ruby /home/oz/bin/fusuma -v
21107 oz         20   0 12976  3820  3520 S  0.0  0.0  0:09.52 libinput-debug-events --device /dev/input/event12
```

iberianpig, 2019-01-15:
Thanks for reporting this issue.

```
D, [2019-01-14T23:43:53.906459 #21097] DEBUG -- : {:time=>"+7678.68s", :event=>"GESTURE_PINCH_END", :finger=>"3", :directions=>{:move=>{:x=>"cancelled", :y=>nil}, :zoom=>nil}}
```
In this line the libinput parsing seems to have failed.

> Continues working for 10 mins, after that it stops triggering actions on mouse events

Do you need a mouse to reproduce this issue?

ozym4nd145, 2019-01-23:
Here's the output for `fusuma -l`
```
$ fusuma -l
Synaptics s3203_ver5
```

iberianpig, 2019-01-23:
Thanks.

Could you try following command if stopping recognition of gestures?

```
sudo modprobe -r psmouse
sudo modprobe psmouse
```

Ref: https://github.com/iberianpig/fusuma/issues/97#issuecomment-430042834

ozym4nd145, 2019-01-24:
I tried it today. It didn't work.
Specifically, I ran the commands when it was working and it didn't disrupt the normal flow.
Then I tried the above commands when it stopped working, but they didn't get it to start working again.

closed 2019-04-01 by stale[bot]

nhamonin, 2019-12-25:
@ozym4nd145 have u already fix it?

jiaobendaye, 2020-01-20:
i have the same problem

iberianpig, 2020-01-21:
Dld you get any errors in console?

jiaobendaye, 2020-01-21:
> Dld you get any errors in console?

no, the reason seems like is that the fusuma can not recognize four fingers and three fingers well.

0xB00B, 2022-10-04:
i have the same issue, latest manjaro kde. the issue happens after the laptop either sleeps or if i try to use 4 finger gesture.

iberianpig, 2022-10-04:
Could you keep fusuma logs when the error occurred so that I can investigate? I am unable to reproduce it so I need environment information and logs.

tapebysea, 2025-09-15:
I'm currently running into this issue, here's a log of the output with the verbose flag;

[fusuma.log](https://github.com/user-attachments/files/22325280/fusuma.log)

It doesn't seem to produce any errors though. Oddly, it only happens to me when fusuma is executed quickly after starting the window manager. If I put `exec sleep 20 && fusuma` in my sway config it seem to run fine, albeit slow to start up. If fusuma stops I can kill the process and execute it again from the terminal and it seems to just work fine after that.

here's my config.yml as well in case that's helpful;
```

swipe:
  3:
    up:
      update:
        command: "pactl set-sink-volume \\@DEFAULT_SINK@ +1%"
        interval: 0.1
    down:
      update:
        command: "pactl set-sink-volume \\@DEFAULT_SINK@ -1%"
        interval: 0.1

```