iberianpig/fusuma #113: Gestures take a few seconds to react

wonx, 2019-01-16:
Usually fusuma works as intended, reacting to gestures immediately, but sometimes it consistently takes a few seconds (2-3s) to execute the command. It doesn't matter if the command is ran at startup or manually in a console. Moreover, the whole OS (Ubuntu 18.04 in my case) freezes while the gesture waits to be executed (even the cursor stops blinking).

**To Reproduce**
Steps to reproduce the behavior:
1. Do a gesture.
2. Wait...
3. The gesture is executed.

**Expected behavior**
The gesture should be executed immediately, in a seamless manner.

**Versions**
```
I, [2019-01-16T03:31:05.683477 #29301]  INFO -- : ---------------------------------------------
I, [2019-01-16T03:31:05.683551 #29301]  INFO -- : Fusuma: 0.10.2
I, [2019-01-16T03:31:05.686091 #29301]  INFO -- : libinput: 1.10.4
I, [2019-01-16T03:31:05.687169 #29301]  INFO -- : OS: Linux 4.15.0-43-generic #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018
I, [2019-01-16T03:31:05.688534 #29301]  INFO -- : Distribution: Ubuntu 18.04.1 LTS \n \l
I, [2019-01-16T03:31:05.689534 #29301]  INFO -- : Desktop session: ubuntu
I, [2019-01-16T03:31:05.689585 #29301]  INFO -- : ---------------------------------------------

```

**Additional context**
Any additional context, your `~/.config/fusuma/config.yml` or data that might be necessary to reproduce the issue.
```
:~$ cat ~/.config/fusuma/config.yml
swipe:
  3: 
    left: 
      command: 'xdotool key alt+Right'
    right: 
      command: 'xdotool key alt+Left'
    up: 
      command: 'xdotool key super'
    down: 
      command: 'xdotool key super'
  4:
    left: 
      command: 'xdotool key ctrl+alt+Down'
    right: 
      command: 'xdotool key ctrl+alt+Up'
    up: 
      command: 'xdotool key ctrl+alt+Down'
    down: 
      command: 'xdotool key ctrl+alt+Up'
pinch:
  in:
    command: 'xdotool key ctrl+plus'
  out:
     command: 'xdotool key ctrl+minus'

threshold:
  swipe: 0.4
  pinch: 0.4

interval:
  swipe: 0.8
  pinch: 0.1

```

wonx, 2019-01-16:
Found the issue, as described in this bug report: https://github.com/bulletmark/libinput-gestures/issues/151

It turns out it has to do with  xdotool and having multiple keyboard layouts. The issue can be easily reproduced by typing _xdotool key super_ or _xdotool type "aaaaaaaaaaaaaaa"_ (the more letters, the longer it takes to process). Moreover, for some reason, the delay only occurs with one of the two layouts, the one which is not the default one (the one not on top in "input sources").

iberianpig, 2019-01-19:
This issue seems to be caused by "xdootool", not "fusuma".
We should consider another way instead of xdotool.

For example, to switch to next or previous workspace, set fusuma to following script (as `~/desktop_switcher.sh`)

```desktop_switcher.sh
#!/usr/bin/env bash

if [ $# -gt 1 ]; then
  echo "expect 1 argument" 1>&2
  exit 1
fi

switch_number=1
if [ $# -eq 1 ]; then
  switch_number=$1
fi

current_workspace=$(wmctrl -d | grep '\*' | cut -d' ' -f1)

cmd="wmctrl -s"
to=$((current_workspace + switch_number))
if [ $to -lt 0 ]; then
  exit 0
fi
$cmd $to

```

~/.config/fusuma/config.yml
```
swipe:
  4:
    left:
      command: 'sh ~/Dropbox/tool/linux/script/desktop_switcher.sh 1'
    right:
      command: 'sh ~/Dropbox/tool/linux/script/desktop_switcher.sh -1' 
```

iberianpig, 2019-03-21:
If I could make native extension for xdotool, this issue may be solved. 
Ref https://github.com/Hikari9/comfortable-swipe/issues/30#issuecomment-462688355

Ospeto, 2019-03-31:
> Found the issue, as described in this bug report: [bulletmark/libinput-gestures#151](https://github.com/bulletmark/libinput-gestures/issues/151)
> 
> It turns out it has to do with xdotool and having multiple keyboard layouts. The issue can be easily reproduced by typing _xdotool key super_ or _xdotool type "aaaaaaaaaaaaaaa"_ (the more letters, the longer it takes to process). Moreover, for some reason, the delay only occurs with one of the two layouts, the one which is not the default one (the one not on top in "input sources").

where should i type xdotool key super? i need to use two keyboard layout , help me

vinyoliver, 2019-04-10:
Any news on this fix?

iberianpig, 2019-04-10:
No, not yet. I haven't had a time for working on fix issues these days😭

epakai, 2019-05-17:
I did some changes to prevent fusuma hanging when starting long running process. I also had issue with children dying if fusuma is killed (probably not an issue in normal use, but was annoying when testing).

I think as a side effect this change would also eliminate delay issues for xdotool commands that take longer than normal. 
epakai@a408bd9ffe21ce74e2d82971a211eb25cddfb638

I'm not familiar with ruby, but hopefully I didn't miss anything fundamental.

iberianpig, 2019-05-18:
Thank you very much. 
This change looks good.
Could you make PR?

iberianpig, 2019-05-24:
@wonx @vinyoliver 
Could you confirm that this issue is fixed?

- `gem update fusuma`
- `pkill fusuma -f`
- restart fusuma on your terminal.

podgorny, 2019-05-25:
Same problem - the  2nd, not default layout causes delay.

iberianpig, 2019-05-25:
😢

iberianpig, 2019-05-25:
It may be better to handle libxdo from native extensions.

wonx, 2019-05-31:
> @wonx @vinyoliver
> Could you confirm that this issue is fixed?
> 
> * `gem update fusuma`
> * `pkill fusuma -f`
> * restart fusuma on your terminal.

Sorry, I didn't see your answer until now. I updated fusuma by running these commands, but the delay when using the non default keyboard layout is still present.

vinyoliver, 2019-06-01:
> @wonx @vinyoliver
> Could you confirm that this issue is fixed?
> 
> * `gem update fusuma`
> * `pkill fusuma -f`
> * restart fusuma on your terminal.

sorry for the late response. The delay is still there. :(

iberianpig, 2019-06-02:
Thank you for confirmation.

I wanted to confirm if this change worked.

> https://github.com/iberianpig/fusuma/issues/113#issuecomment-493388089

It seems that this problem has not been solved yet.

By the way, is this problem a problem with Gnome Shell?
(I'm using Pantheon on elementary OS, but it has not occurred the problem)
If so, I'll install Gnome Shell and try to reproduce.

Saikorox, 2019-06-02:
@iberianpig 
Using ubuntu/gnome - problem is still there after latest update. Don't know about others.

danilyanich, 2019-07-09:
I wrote simple window switcher based on https://github.com/iberianpig/fusuma/issues/113#issuecomment-455750664 (It replicates alt-tab). Maybe someone will find it useful.
```bash
#!/usr/bin/env bash
set -euxo pipefail

if [ $# -gt 1 ]; then
  echo "expect 1 argument" 1>&2
  exit 1
fi

switch_number=1
if [ $# -eq 1 ]; then
  switch_number=$1
fi

current_workspace=$(wmctrl -d | grep '\*' | cut -d' ' -f1)
active_window=$(wmctrl -a :ACTIVE: -v 2>&1 | grep 'Using' | cut -d' ' -f3)
windows_in_workspace=$(wmctrl -lx | awk -v num="$current_workspace" '$2==num {print $1}')

cmd="wmctrl -i -a"

if [ $switch_number -eq "1" ];then
  switch_to_window=$(echo $windows_in_workspace | sed s/.*$active_window// | awk '{print $1}')

  if [ -z "$switch_to_window" ];then
     switch_to_window=$(echo $windows_in_workspace | awk '{print $1}')
  fi

  $cmd $switch_to_window
  exit 0
fi

if [ $switch_number -eq "-1" ];then
  switch_to_window=$(echo $windows_in_workspace | sed s/$active_window.*// | awk '{print $NF}')

  if [ -z "$switch_to_window" ];then
     switch_to_window=$(echo $windows_in_workspace | awk '{print $NF}')
  fi

  $cmd $switch_to_window
  exit 0
fi

exit 1
```
Usage is the same as in original script.

iberianpig, 2019-08-14:
ref https://github.com/bulletmark/libinput-gestures/issues/151#issuecomment-520750300

ulandj, 2019-09-01:
> ref [bulletmark/libinput-gestures#151 (comment)](https://github.com/bulletmark/libinput-gestures/issues/151#issuecomment-520750300)

Example which works fine for me:
`swipe:
  3:
    left:
      command: 'xte "keydown Alt_L" "key Tab" "keyup Alt_L"'
    right:
      command: 'xte "keydown Alt_L" "keydown Shift_L" "key Tab" "keyup Alt_L" "keyup Shift_L"'
    up:
      command: 'xte "keydown Alt_L" "keydown Control_L" "key Down" "keyup Control_L" "keyup Alt_L"'
    down:
      command: 'xte "keydown Alt_L" "keydown Control_L" "key Up" "keyup Control_L" "keyup Alt_L"'
  4:
    up:
      command: "dbus-send --session --type=method_call --dest=org.gnome.Shell /org/gnome/Shell org.gnome.Shell.Eval string:'Main.overview.toggle();'"
    down:
      command: 'xte "keydown Super_L" "key A" "keyup Super_L"'
pinch:
  in:
    command: 'xte "keydown Control_L" "key minus" "keyup Control_L"'
  out:
    command: 'xte "keydown Control_L" "key plus" "keyup Control_L"'

threshold:
  swipe: 0.8
  pinch: 0.8

interval:
  swipe: 0.8
  pinch: 0.1
`

ghilman27, 2019-10-02:
I think this issue still hasn't been resolved? Been using `xte` and manual script replacing `xdotool` but the problem still exists. Sometimes can't swipe and need to wait seconds to work again

iberianpig, 2019-10-13:
I released [fusuma-plugin-wmctrl](https://github.com/iberianpig/fusuma-plugin-wmctrl).
This plugin helps switching workspace without xdotool.

pedro-lb, 2019-10-16:
> I released [fusuma-plugin-wmctrl](https://github.com/iberianpig/fusuma-plugin-wmctrl).
> This plugin helps switching workspace without xdotool.

Thanks @iberianpig ! Works really well.

wonx, 2019-10-18:
How does it work? I installed it according to the instructions, restarted fusuma, but the delay is still there. 

I checked if the plugin was loaded with fusuma, and apparently it does. But I don't see a difference.

```
D, [2019-10-18T15:41:19.823711 #16479] DEBUG -- : Enabled Plugins: 
D, [2019-10-18T15:41:19.823796 #16479] DEBUG -- :   Fusuma::Plugin::Buffers::GestureBuffer
D, [2019-10-18T15:41:19.823809 #16479] DEBUG -- :   Fusuma::Plugin::Detectors::PinchDetector
D, [2019-10-18T15:41:19.823818 #16479] DEBUG -- :   Fusuma::Plugin::Detectors::RotateDetector
D, [2019-10-18T15:41:19.823829 #16479] DEBUG -- :   Fusuma::Plugin::Detectors::SwipeDetector
D, [2019-10-18T15:41:19.823837 #16479] DEBUG -- :   Fusuma::Plugin::Events::Records::GestureRecord
D, [2019-10-18T15:41:19.823846 #16479] DEBUG -- :   Fusuma::Plugin::Events::Records::IndexRecord
D, [2019-10-18T15:41:19.823860 #16479] DEBUG -- :   Fusuma::Plugin::Events::Records::TextRecord
D, [2019-10-18T15:41:19.823916 #16479] DEBUG -- :   Fusuma::Plugin::Executors::CommandExecutor
D, [2019-10-18T15:41:19.823926 #16479] DEBUG -- :   Fusuma::Plugin::Executors::WmctrlExecutor
D, [2019-10-18T15:41:19.823935 #16479] DEBUG -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
D, [2019-10-18T15:41:19.823946 #16479] DEBUG -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
D, [2019-10-18T15:41:19.823955 #16479] DEBUG -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
```

iberianpig, 2019-10-19:
Did you set workspace parameters?https://github.com/iberianpig/fusuma-plugin-wmctrl/blob/master/README.md#example

This plugin use wmctl Instead of xdotool, and wmctl can switch current workspace.

This is not perfect, but it can be useful for switching workspaces.

wonx, 2019-10-19:
Ahh, ok, it was that, it switches workspaces for now. Thanks

wonx, 2019-12-18:
I mean, the bug is still open, and I think it's quite relevant, but it's xdotool's problem, not fusuma's.

pedro-lb, 2020-02-18:
Not stale

iberianpig, 2020-03-02:
Please try this plugin https://github.com/iberianpig/fusuma-plugin-sendkey

illright, 2020-05-19:
Here's [a nice Python script](https://askubuntu.com/a/873943/601580) that allows using `wmctrl` to cycle through windows. A nice way to avoid `xdotool` if you only need Alt+Tab functionality.

closed 2020-07-25 by stale[bot]

HideyoshiNakazone, 2020-12-14:
> Found the issue, as described in this bug report: [bulletmark/libinput-gestures#151](https://github.com/bulletmark/libinput-gestures/issues/151)
>
> It turns out it has to do with xdotool and having [...]

For that problem i suggest using xte from xautomation package.

iberianpig, 2020-12-14:
I suggest https://github.com/iberianpig/fusuma-plugin-sendkey instead of xdotool.
This plugin has no dependency to X11.