iberianpig/fusuma #115: Hey, I've created a native alternative to this software

Coffee2CodeNL, 2019-02-15:
Hey there :grin: 

I've written an alternative to this software that implements libinput natively, instead of having to rely on unstable tools like `libinput debug-events`

I have also referenced your project in my repository!

I hope this has piqued your interest!

[Link to my repo](https://github.com/Coffee2CodeNL/gebaar-libinput)

iberianpig, 2019-02-16:
I saw your project!

I am interested in how performance improvement which is bigger than libinput-debug-events.

If it seems to be effective, fusuma will support native input of libinput as a plugin😎

closed 2019-02-16 by iberianpig

jirutka, 2022-04-04:
@Coffee2CodeNL, it’s sad that this project of yours is dead. :(

Coffee2CodeNL, 2022-04-19:
> @Coffee2CodeNL, it’s sad that this project of yours is dead. :(

It's not dead per se, I just have been dealing with some personal issues.