iberianpig/fusuma #118: Apple magic mouse 1 support

4sStylZ, 2019-04-18:
Hi,

**Is your feature request related to a problem? Please describe.**
I would love to use my magic mouse 1 (there is two different device) with some gesture like on MacOS.

**Describe the solution you'd like**
I want to swipe and map some function on the 2 ~ 3 ~ 4 fingers swipe (and tap, but I know the issue on libinput who don’t provide tap informations)

**Describe alternatives you've considered**
touchegg, but he is not working on my elementary OS, even on my touchpad.

**Additional context**
I use a Zenbook asus with 
* ElementaryOS 5.0 Juno 64 bit
* Gtk+ 3.22.30

I have some information about what is interpreted when I swipe on my magic mouse : 

```
$ libinput-debug-events
WARNING: This compatibility wrapper will be removed in the future. Please run 'libinput debug-events' instead
-event3   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k
-event8   DEVICE_ADDED     Asus Wireless Radio Control       seat0 default group2  cap:k
-event5   DEVICE_ADDED     Video Bus                         seat0 default group3  cap:k
-event0   DEVICE_ADDED     Lid Switch                        seat0 default group4  cap:S
-event2   DEVICE_ADDED     Power Button                      seat0 default group5  cap:k
-event1   DEVICE_ADDED     Sleep Button                      seat0 default group6  cap:k
-event19  DEVICE_ADDED     Kensington      Kensington Expert Mouse seat0 default group7  cap:p left scroll-nat scroll-button
-event7   DEVICE_ADDED     TypeMatrix.com USB Keyboard       seat0 default group8  cap:k
-event18  DEVICE_ADDED     TypeMatrix.com USB Keyboard       seat0 default group8  cap:kp scroll-nat
-event9   DEVICE_ADDED     USB2.0 HD UVC WebCam: USB2.0 HD   seat0 default group9  cap:k
-event6   DEVICE_ADDED     Souris de christine               seat0 default group10 cap:p left scroll-nat scroll-button
-event11  DEVICE_ADDED     ELAN1200:00 04F3:3059 Touchpad    seat0 default group11 cap:pg  size 103x69mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on
-event12  DEVICE_ADDED     HDA Intel PCH Headphone           seat0 default group12 cap:
-event13  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=3       seat0 default group12 cap:
-event14  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=7       seat0 default group12 cap:
-event15  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=8       seat0 default group12 cap:
-event16  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=9       seat0 default group12 cap:
-event17  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=10      seat0 default group12 cap:
-event10  DEVICE_ADDED     Asus WMI hotkeys                  seat0 default group13 cap:k
-event4   DEVICE_ADDED     AT Translated Set 2 keyboard      seat0 default group14 cap:k
-event6   POINTER_MOTION    +1.31s	  0.00/  0.23
 event6   POINTER_MOTION    +1.33s	  4.08/  2.72
 event6   POINTER_MOTION    +1.34s	  8.04/  4.82
 event6   POINTER_MOTION    +1.35s	  8.68/  5.21
 event6   POINTER_MOTION    +1.36s	  5.22/  4.35
 event6   POINTER_MOTION    +1.37s	  4.18/  0.84
 event6   POINTER_MOTION    +1.74s	  0.00/ -0.69
 event6   POINTER_AXIS      +2.29s	vert 15.00* horiz 0.00 (wheel)
 event6   POINTER_AXIS      +2.53s	vert 15.00* horiz 0.00 (wheel)
 event6   POINTER_AXIS      +2.91s	vert 0.00 horiz -15.00* (wheel)
```

The logs reported demonstrate that the mouse is recognized and also that moving my 3 fingers on it is detected by libinput.

When I use it on Fusuma, nothing is prompted but he detect the mouse.

```
$ fusuma --list-devices
Kensington      Kensington Expert Mouse
TypeMatrix.com USB Keyboard
Apple Mouse
ELAN1200:00 04F3:3059 Touchpad
```

Any suggestion ?

I don’t see any support for this mouse on Ubuntu since the 16/04 wich is very sad…

iberianpig, 2019-05-05:
```
 event6   POINTER_AXIS      +2.29s	vert 15.00* horiz 0.00 (wheel)
 event6   POINTER_AXIS      +2.53s	vert 15.00* horiz 0.00 (wheel)
 event6   POINTER_AXIS      +2.91s	vert 0.00 horiz -15.00* (wheel)
```

Yes, It seems that magic mouse 1 reports that `POINTER_AXIS` detects as a scroll(wheel).

So, it works well if fusuma parse `POINTER_AXIS` as a swipe gesture event.
However, it needs to makes a patch for magic mouse1.

4sStylZ, 2019-05-09:
Hi, 
Thank for the support.
If you want to make the patch and you don’t have a Magic mouse for testing :
 - I can lend you mine for testing purpose. 
 - More simple : I can also help you to test it / debug it. We can planify a meet on Teamspeak 3 or discord and I can also share my screen.

Seeya

iberianpig, 2019-05-11:
Hi, 
Thanks for your kindness.

My friend lends me magic mouse 1 from.
So I can debug with it😆

closed 2019-10-09 by stale[bot]

BenoitDuffez, 2019-12-16:
Hi, same here (except on a HP laptop). I'm using Elementary OS and touchégg does not work, however fusuma works well.

The main issue is that it's working fine for my touchpad, but I rarely use it. I rather use my magic mouse (1) but horizontal swipe does not trigger anything on fusuma. What I mean by that is whatever I do with finger touches/swipes on the mouse only trigger "POINTER_MOTION" on fusuma. When I do the same on the touchpad, it correctly shows the number of fingers detected and it's GESTURE_SWIPE_XXX.

What can I do to help? I'd like specifically to have 1 and 2 finger swipe gestures.

iberianpig, 2019-12-16:
Hi, @BenoitDuffez 

Does it have a different of `libinput debug-events` outputs when moving Magic Mouse and when swiping 1 or 2 fingers on Magic Mouse?

BenoitDuffez, 2019-12-17:
Nope, exactly the same. I'll post a log tomorrow.

BenoitDuffez, 2020-09-25:
Well, that was a long night.

```
# libinput-debug-events
WARNING: This compatibility wrapper will be removed in the future. Please run 'libinput debug-events' instead
-event2   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k
-event4   DEVICE_ADDED     Video Bus                         seat0 default group2  cap:k
-event1   DEVICE_ADDED     Lid Switch                        seat0 default group3  cap:S
-event0   DEVICE_ADDED     Sleep Button                      seat0 default group4  cap:k
-event5   DEVICE_ADDED     G-Tech Wireless Dongle            seat0 default group5  cap:kp left scroll-nat scroll-button
-event22  DEVICE_ADDED     Logitech K750                     seat0 default group6  cap:kp scroll-nat
-event23  DEVICE_ADDED     Logitech K750                     seat0 default group7  cap:kp scroll-nat
-event26  DEVICE_ADDED     Logitech K750                     seat0 default group6  cap:kp scroll-nat
-event24  DEVICE_ADDED     Logitech M515                     seat0 default group8  cap:kp left scroll-nat scroll-button
-event25  DEVICE_ADDED     Logitech K750                     seat0 default group7  cap:kp scroll-nat
-event6   DEVICE_ADDED     GooglęPixel USB-C earbudsꤳ赀༢ީ鮍蔈굋࢙ Pixel USB-C earbuds seat0 default group9  cap:k
-event27  DEVICE_ADDED     Souris de bicou                   seat0 default group10 cap:p left scroll-nat scroll-button
-event18  DEVICE_ADDED     HP HD Camera: HP HD Camera        seat0 default group11 cap:k
-event9   DEVICE_ADDED     HDA Intel PCH Mic                 seat0 default group12 cap:
-event10  DEVICE_ADDED     HDA Intel PCH Line                seat0 default group12 cap:
-event11  DEVICE_ADDED     HDA Intel PCH Dock Line Out       seat0 default group12 cap:
-event12  DEVICE_ADDED     HDA Intel PCH Headphone           seat0 default group12 cap:
-event13  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=3       seat0 default group12 cap:
-event14  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=7       seat0 default group12 cap:
-event15  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=8       seat0 default group12 cap:
-event16  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=9       seat0 default group12 cap:
-event17  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=10      seat0 default group12 cap:
-event3   DEVICE_ADDED     AT Translated Set 2 keyboard      seat0 default group13 cap:k
-event7   DEVICE_ADDED     SynPS/2 Synaptics TouchPad        seat0 default group14 cap:pg  size 99x45mm tap(dl off) left scroll-nat scroll-2fg-edge dwt-on
-event8   DEVICE_ADDED     HP Wireless hotkeys               seat0 default group15 cap:k
-event19  DEVICE_ADDED     HP WMI hotkeys                    seat0 default group16 cap:k
-event21  DEVICE_ADDED     ACPI Virtual Keyboard Device      seat0 default group17 cap:k
-event27  POINTER_AXIS      +2.78s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +2.84s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +2.86s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +2.88s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +2.90s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +2.92s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +2.94s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +2.96s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +2.98s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +3.01s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +3.03s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +3.05s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +3.08s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +3.17s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +4.87s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +4.92s      vert 0.00 horiz -30.00* (wheel)
 event27  POINTER_AXIS      +4.95s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +4.96s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +4.99s      vert 0.00 horiz -30.00* (wheel)
 event27  POINTER_AXIS      +5.01s      vert 0.00 horiz -30.00* (wheel)
 event27  POINTER_AXIS      +5.04s      vert 0.00 horiz -30.00* (wheel)
 event27  POINTER_AXIS      +5.08s      vert 0.00 horiz -30.00* (wheel)
 event27  POINTER_AXIS      +5.12s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +5.16s      vert 0.00 horiz -15.00* (wheel)
 event27  POINTER_AXIS      +5.23s      vert 0.00 horiz -15.00* (wheel)
```

First move (3sec) is one finger horizontal swipe from right to left.
Second move (5sec) is two fingers horizontal swipe from right to left.

This is the same (with a 3rd test with 3 fingers swipe) using the internal trackpad:

```
 event7   POINTER_MOTION    +3.29s       -0.78/  0.41
 event7   POINTER_MOTION    +3.31s       -5.20/  1.09
 event7   POINTER_MOTION    +3.32s       -4.59/  1.55
 event7   POINTER_MOTION    +3.35s       -9.61/  1.74
 event7   POINTER_MOTION    +3.36s       -4.64/ -0.17
[... 43 lines cut ...]
 event7   GESTURE_SWIPE_UPDATE  +8.67s  3 -8.32/ 0.00 (-30.42/ 0.00 unaccelerated)
 event7   GESTURE_SWIPE_UPDATE  +8.72s  3 -14.35/-0.34 (-39.37/-0.94 unaccelerated)
 event7   GESTURE_SWIPE_UPDATE  +8.75s  3 -7.29/-0.35 (-19.69/-0.94 unaccelerated)
 event7   GESTURE_SWIPE_UPDATE  +8.77s  3 -4.64/-0.35 (-12.53/-0.94 unaccelerated)
[... cut ...]
 event7   GESTURE_SWIPE_UPDATE  +9.23s  3 -5.30/ 0.00 (-14.32/ 0.00 unaccelerated)
 event7   GESTURE_SWIPE_UPDATE  +9.26s  3 -3.97/ 0.00 (-10.74/ 0.00 unaccelerated)
 event7   GESTURE_SWIPE_UPDATE  +9.29s  3 -3.31/ 0.00 (-8.95/ 0.00 unaccelerated)
 event7   GESTURE_SWIPE_UPDATE  +9.32s  3 -3.31/ 0.00 (-8.95/ 0.00 unaccelerated)
 event7   GESTURE_SWIPE_UPDATE  +9.36s  3 -2.65/ 0.00 (-7.16/ 0.00 unaccelerated)
 event7   GESTURE_SWIPE_END  +9.40s     3
```

So the problem seems to be on libinput, not fusuma as it seems... I'm confused now :)

iberianpig, 2020-09-26:
It seems that Libinput cannot detect 1,2 fingers swipes on the magic mouse, and Libinput detects as a wheel.
Fusuma converts stdout of `libinput debug-events` to user commands.
So I'm difficult to fix this issue.

BenoitDuffez, 2020-09-26:
I agree, the problem comes from libinput. Do you have any idea how to fix this?