iberianpig/fusuma #119: Crashes

uguntde, 2019-05-12:
I am using fusuma on a DELL XPS laptop and ubuntu 19.04. I see regular crashes with the following error message:

I, [2019-05-12T21:49:53.792524 #6674]  INFO -- : Execute: xdotool key alt+Right
Traceback (most recent call last):
	22: from /usr/local/bin/fusuma:23:in `<main>'
	21: from /usr/local/bin/fusuma:23:in `load'
	20: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/exe/fusuma:40:in `<top (required)>'
	19: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma.rb:24:in `run'
	18: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma.rb:76:in `run'
	17: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/libinput_commands.rb:38:in `debug_events'
	16: from /usr/lib/ruby/2.5.0/open3.rb:95:in `popen3'
	15: from /usr/lib/ruby/2.5.0/open3.rb:205:in `popen_run'
	14: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/libinput_commands.rb:39:in `block in debug_events'
	13: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/libinput_commands.rb:39:in `each'
	12: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/libinput_commands.rb:39:in `block (2 levels) in debug_events'
	11: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma.rb:80:in `block in run'
	10: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/event_stack.rb:15:in `generate_command_executor'
	 9: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/pinch.rb:22:in `enough?'
	 8: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/pinch.rb:29:in `enough_diameter?'
	 7: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/pinch.rb:43:in `threshold'
	 6: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/config.rb:18:in `threshold'
	 5: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/config.rb:57:in `threshold'
	 4: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/config.rb:71:in `search_config_cached'
	 3: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/config.rb:119:in `cache'
	 2: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/config.rb:71:in `block in search_config_cached'
	 1: from /var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/config.rb:82:in `search_config'
/var/lib/gems/2.5.0/gems/fusuma-0.10.2/lib/fusuma/config.rb:80:in `search_config': undefined method `[]' for 0.5:Float (NoMethodError)

iberianpig, 2019-05-13:
Could you paste output of `fusuma --version` and your `~/config.yml`?

uguntde, 2019-05-13:
Here they are:

fusuma --version
I, [2019-05-13T05:46:48.401667 #28217]  INFO -- : 
---------------------------------------------
I, [2019-05-13T05:46:48.401797 #28217]  INFO -- : Fusuma: 0.10.2
I, [2019-05-13T05:46:48.407593 #28217]  INFO -- : libinput: 1.12.6
I, [2019-05-13T05:46:48.409289 #28217]  INFO -- : OS: Linux 
5.0.0-13-generic #14-Ubuntu SMP Mon Apr 15 14:59:14 UTC 2019
I, [2019-05-13T05:46:48.461955 #28217]  INFO -- : Distribution: Ubuntu 
19.04 \n \l
I, [2019-05-13T05:46:48.475319 #28217]  INFO -- : Desktop session: 
ubuntu-wayland
I, [2019-05-13T05:46:48.475371 #28217]  INFO -- : 
---------------------------------------------

~/.config/fusuma$ cat config.yml
swipe:
   3:
     left:
       shortcut: 'alt+Left'
     right:
       shortcut: 'alt+Right'
     up:
       shortcut: 'alt+F1'
     down:
       shortcut: 'ctrl+w'
   4:
     left:
       shortcut: 'ctrl+alt+Left'
     right:
       shortcut: 'ctrl+alt+Right'
     up:
       shortcut: 'ctrl+alt+Down'
     down:
       shortcut: 'ctrl+alt+Up'
pinch:
   in:
     shortcut: 'ctrl+plus'
   out:
      shortcut: 'ctrl+minus'

threshold:
   swipe: 0.2
   pinch: 0.5

interval:
   swipe: 0.5
pinch: 0.5

uguntde, 2019-05-14:
This seems to happen when I pinch - maybe this helps to debug it. Often I seem to pinch without intention. 

I have now increased 

interval:
  swipe: 0.8
  pinch: 0.8

and it hasn't happened since.

What does interval change?

iberianpig, 2019-05-24:
Yes, and I noticed that there was no indent at the following section.

```yaml
interval:
   swipe: 0.5
pinch: 0.5  # <--- need more space!
```

iberianpig, 2019-05-24:
> What does interval change?

You can change the execution interval between commands.
For example, it is effective when you want to zoom in by pinching in the screen.
By reducing the interval between consecutive gesture events, you can execute commands more smoothly.

closed 2019-08-02 by stale[bot]