iberianpig/fusuma #125: Fusuma must be started after connecting bluetooth trackpad

wonx, 2019-06-04:
**Describe the bug**
I am not really sure if it's a bug or not.

Gestures in bluetooth trackpads (like Apple's Magic trackpad) are not recognized if fusuma is started before the device is paired.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch fusuma as a daemon (fusuma -d) when the system starts.
2. Connect a bluetooth trackpad. Notice how the bluetooth trackpad works, but not the gestures.
3. Run `sudo killall fusuma` and `fusuma` once again.
4. Notice how the gestures now work on the bluetooth trackpad.

**Expected behavior**
fusuma should detect new trackpad devices connected after it has been launched.

**Versions**
You can get this information from copy and paste the output of `fusuma --version` from the command line. Also, please include the OS and what version of the OS you're running.

```
user@TM1703:~$ fusuma --version
I, [2019-06-04T17:04:37.677538 #3533]  INFO -- : ---------------------------------------------
I, [2019-06-04T17:04:37.677576 #3533]  INFO -- : Fusuma: 0.11.1
I, [2019-06-04T17:04:37.679002 #3533]  INFO -- : libinput: 1.10.4
I, [2019-06-04T17:04:37.679880 #3533]  INFO -- : OS: Linux 4.15.0-50-generic #54-Ubuntu SMP Mon May 6 18:46:08 UTC 2019
I, [2019-06-04T17:04:37.680549 #3533]  INFO -- : Distribution: Ubuntu 18.04.2 LTS \n \l
I, [2019-06-04T17:04:37.681095 #3533]  INFO -- : Desktop session: ubuntu
I, [2019-06-04T17:04:37.681118 #3533]  INFO -- : ---------------------------------------------

```

Natetronn, 2019-06-07:
I believe I'm running into this as well. I have a Apple Magic Trackpad and it doesn't connect and function like a Trackpad normally would until after I click it a couple times. After this it initializes and connects to the bluetooth and it starts to function normally. This happens after every restart. Fusuma command doesn't seem to work, unless I manually ru `fusuma -d` directly in the terminal every time.

wonx, 2019-06-10:
I have fusuma to launch when I start my session, so what I usually do, is make sure the trackpad connected at the login screen. Otherwise, I have to manually restart the service.

iberianpig, 2019-06-12:
Could you check `libinput debug-events` detect the event to connect a new device?
If fusuma can catch the event, we should implement it to reload fusuma only when detecting a new device.

wonx, 2019-06-14:
Yes, `libinput debug-events` detects when the trackpad is connected for the first time. It shows the following:

`-event18  DEVICE_ADDED     Apple Wireless Trackpad           seat0 default group13 cap:pg  size 132x112mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfing`

Natetronn, 2019-06-14:
I get something similar:

`-event19  DEVICE_ADDED     {MY NAME} Trackpad           seat0 default group7  cap:pg  size 132x112mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger`

wonx, 2019-10-07:
Can it be reopened?

Natetronn, 2019-10-14:
I setup an alias: `alias swipe="fusuma -d"` Not ideal but, gets the job done.

iberianpig, 2019-10-16:
It seems to be the same issue.

https://github.com/bulletmark/libinput-gestures/issues/228

iberianpig, 2019-10-16:
I will implement this feature in `lib/fusuma/plugin/filters/libinput_device_filter.rb`.

I think that fusuma should reload `@keep_device_ids` when it detects `DEVICE_ADDED` in the record.

closed 2019-10-19 by iberianpig

iberianpig, 2019-10-19:
@wonx @Natetronn Please try version 1.2.1.
If you have a any problem, please reopen.

Natetronn, 2019-10-19:
Thank you for this!

That seems to have addressed it for me. I restarted the computer, after login I clicked on my touchpad to get it to register, trackpad starts working with mouse movements, opened a website and navigated through a couple of pages, then I swiped both ways and it worked fine. I have fusuma -d in startup applications, as was the case before.

I'll give it a few days of playing around to fully confirm but, on first go, it seems to be working.

Natetronn, 2019-10-19:
P.s. I ran `sudo gem install fusuma` to update but, not sure that was the proper way.

commit 00fbdba2 (2019-10-19): Reload KeepDevice in Filter::LibinputDeviceFilter when DEVICE_ADDED is detected

* remove Device.given_devices=, Device.narrow_available_devices
* add Plugin::Filters::LibinputDeviceFilter::KeepDevice

Closes #125