iberianpig/fusuma #130: 'Touchpad is not found'

RobinZ1, 2019-07-29:
E, [2019-07-29T15:15:20.612207 #12220] ERROR -- : Touchpad is not found

Dear Contributor,
this issue occurs when fusuma is launched

iberianpig, 2019-07-29:
Hi RobinZ1,

Could you tell about your environment? 

Please paste the output of `fusuma --version` from the command line.

dwilmer, 2019-08-03:
I just had this same issue. Here's my version info:
```
I, [2019-08-03T22:23:29.149046 #6856]  INFO -- : ---------------------------------------------
I, [2019-08-03T22:23:29.149088 #6856]  INFO -- : Fusuma: 0.11.1
I, [2019-08-03T22:23:29.150329 #6856]  INFO -- : libinput: 1.10.4
I, [2019-08-03T22:23:29.150888 #6856]  INFO -- : OS: Linux 5.0.0-23-generic #24~18.04.1-Ubuntu SMP Mon Jul 29 16:12:28 UTC 2019
I, [2019-08-03T22:23:29.158708 #6856]  INFO -- : Distribution: Ubuntu 18.04.2 LTS \n \l
I, [2019-08-03T22:23:29.159733 #6856]  INFO -- : Desktop session: ubuntu
I, [2019-08-03T22:23:29.159787 #6856]  INFO -- : ---------------------------------------------
```

dwilmer, 2019-08-03:
Although it does seem to run when I run `sudo fusuma`, maybe that is the solution? I do get another solution, but will try to figure that out / post another issue if it is one.

iberianpig, 2019-08-05:
Please read README.md.
https://github.com/iberianpig/fusuma/blob/master/README.md#1-grant-permission-to-read-the-touchpad-device

> You MUST LOGOUT/LOGIN or REBOOT to assign this group.

Fusuma needs access privileges on the input group, "sudo" just bypass it with root access.

dwilmer, 2019-08-06:
I did logout and login, not realizing that it might not be enough and that a complete reboot was needed as well. I guess I misread the readme, sorry.
@RobinZ1 did you do the same thing I did, or does a reboot not fix the issue for you?

closed 2019-10-12 by stale[bot]