iberianpig/fusuma #132: Fusuma doesnt work in firefox

joe23reptiles, 2019-09-07:
hi- first off: thanks for your awesome work- really enjoy it.

now to my problem. gestures dont work in firefox. they did before and still work everywhere else. havent changed anything. there was no update today or yesterday. no config changed. it just started out of no where not to work anymore when using firefox.

i use manjaro with kde and have set 3 finger swipe gestures. left and right to switch one desktop to the left or right and 3 finger up to show all open applications. 

i dont know how one would be able to recreate the bug. 

this is my config:

swipe:
  3:
    left:
      command: 'xdotool key super+Left'
    right:
      command: 'xdotool key super+Right'
    up:
      command: 'xdotool key super+Up'
    
 

pinch:
  2:
    in:
      command: 'xdotool key ctrl+plus'
      threshold: 0.1
    out:
      command: 'xdotool key ctrl+minus'
      threshold: 0.1

    

threshold:
  swipe: 1
  pinch: 1

interval:
  swipe: 1
  pinch: 1

so basically on desktop 1 i my filemanager open- on desktop 2 firefox. im on desktop 1 and swipe right and im in firefox. when i now swipe left i stay in firefox. every other program i use works normally. 

maybe uve an idea.

thx in advance

virajvchavan, 2019-09-07:
Can you check your system log as you do the gestures? (syslog)

joe23reptiles, 2019-09-07:
this is the end of the log file with swiping in firefox as one of the last entries (i hope)

Sep 07 14:49:34 matebook tracker-extract[5510]: Could not insert metadata for item "fil>
Sep 07 14:49:34 matebook tracker-extract[5510]: If the error above is recurrent for the>
Sep 07 14:49:45 matebook systemd[936]: tracker-extract.service: Succeeded.
Sep 07 14:49:46 matebook dbus-daemon[956]: [session uid=1000 pid=956] Activating via sy>
Sep 07 14:49:46 matebook systemd[936]: Starting Tracker metadata extractor...
Sep 07 14:49:46 matebook dbus-daemon[956]: [session uid=1000 pid=956] Successfully acti>
Sep 07 14:49:46 matebook systemd[936]: Started Tracker metadata extractor.
Sep 07 14:49:46 matebook tracker-extract[5533]: Could not insert metadata for item "fil>
Sep 07 14:49:46 matebook tracker-extract[5533]: If the error above is recurrent for the>
Sep 07 14:49:48 matebook kwin_x11[1026]: qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadW>
Sep 07 14:49:48 matebook kwin_x11[1026]: qt.qpa.xcb: QXcbConnection: XCB error: 9 (BadD>
Sep 07 14:49:50 matebook polkitd[561]: Registered Authentication Agent for unix-process>
Sep 07 14:49:50 matebook polkitd[561]: Unregistered Authentication Agent for unix-proce>
Sep 07 14:49:57 matebook systemd[936]: tracker-extract.service: Succeeded.
Sep 07 14:49:58 matebook dbus-daemon[956]: [session uid=1000 pid=956] Activating via sy>
Sep 07 14:49:58 matebook systemd[936]: Starting Tracker metadata extractor...
Sep 07 14:49:58 matebook dbus-daemon[956]: [session uid=1000 pid=956] Successfully acti>
Sep 07 14:49:58 matebook systemd[936]: Started Tracker metadata extractor.
Sep 07 14:49:58 matebook tracker-extract[5761]: Could not insert metadata for item "fil>
Sep 07 14:49:58 matebook tracker-extract[5761]: If the error above is recurrent for the>
Sep 07 14:49:58 matebook xdg-desktop-por[1597]: Failed to get application states: GDBus>

joe23reptiles, 2019-09-07:
hmm. weird. now it works again. ive no clue whats going on tbh

closed 2019-11-13 by stale[bot]